qmk/qmk_firmware #6203: Planck EZ sleep and power key problems

zored, 2019-06-27:
Sleep and power keys don't work properly in [Planck EZ](https://github.com/zored/zkf/blob/3ac01191f8ecb4d82442e4daf9bcbf4771a3ba76/example/planck/keymap.c#L2628) despite they are ok on [Ergodox EZ](https://github.com/zored/zkf/blob/3ac01191f8ecb4d82442e4daf9bcbf4771a3ba76/example/ergodox_ez/keymap.c#L2692). [FAQ](https://beta.docs.qmk.fm/faq/faq_keymap#power-keys-arent-working) didn't help. Why could that happen ?

fauxpark, 2019-06-27:
What OS/es are you seeing this on?

It looks like `EXTRAKEY_ENABLE = yes` is present both in your rules.mk files and by default on both boards, so it should just work.

llHoYall, 2019-06-28:
I use dz60rgb-ansi board. It doesn't work KC_SLEP, KC_PWR keys also.
My OS is windows10 and sets `EXTRAKEY_ENABLE=yes`.
I have 1up60hse and 1up60hte boards and these are works well.

fauxpark, 2019-06-28:
I'm seeing a pattern here. DZ60RGB and Planck EZ are both ARM, the rest are AVR. Unfortunately, I don't have any ARM boards...

zored, 2019-06-28:
Thank you for responses!
I tested it on Windows 10 and macOS High Sierra. Ergodox works, Planck doesn't.
Yeah, feels like specific keycode handling on some boards...

drashna, 2019-07-03:
Yeah, it definitely sounds like a AVR vs ARM issue here.  

I suspect that it's an issue with the HID endpoint stuff in ChibiOS (the "HAL" for QMK Firmware), but I'm not familiar enough with that code to fix.

fauxpark, 2019-07-03:
ChibiOS should be using the LUFA HID descriptors, which I *know* work (on AVR)... so the problem is elsewhere.

yiancar, 2019-07-04:
If someone wants to chase this further, a debugger and a post on the chibios forum will get you a long way :)

DrumMasterFat, 2019-07-30:
I am having this same issue with ChibiOS on a DZ65RGB. 
I am 100% new to github and troubleshooting QMK, but have some programming experience and would like to help find a resolution for this issue. I have an instance of MSYS64 set up with QMK. If someone could help me understand the steps for executing a debugger to post on the ChibiOS forum, that would be much appreciated.

zored, 2019-12-26:
I will try to investigate this problem.

Here are some general ideas:
- QMK [uses](https://github.com/qmk/qmk_firmware/tree/master/lib) forks of ChibiOS libs. Therefore [fresh code](https://github.com/mabl/ChibiOS) may contain fixes for this issue.
- New ChibiOS code is incompatible with `chconf.h` in QMK and needs a lot for refactoring.
- There is [ChibiOS forum with bugs](http://www.chibios.com/forum/viewforum.php?f=35&sid=22c21f3e0ca05832bfe92b27c7b831ab). I still don't know what question to ask there.
- Here is my [build log](https://github.com/zored/zkf/issues/13#issue-542367098) with libs that I use.

drashna, 2019-12-27:
@zored There is a PR that updates ChibiOS to 19.1.  It may be worth checking out it out #7597

fauxpark, 2019-12-27:
I would also make sure that all of the keyboard "interfaces" are using the `HidUsb` driver by looking at it in Zadig (but don't install anything with it). It sometimes happens that you accidentally install the bootloader driver onto the keyboard while it's not in bootloader mode. This page should help clear it up: https://docs.qmk.fm/#/driver_installation_zadig?id=recovering-from-installation-to-wrong-device

tzarc, 2020-02-25:
PR #8156 seemed to fix the SLEEP/WAKE/PWR buttons, might be wise retrying to see if it's sorted now.
I did the build with a `future`-based branch, but it applied to master anyway... hopefully it's still good.

DrumMasterFat, 2020-02-25:
I just compiled the latest master build for a DZ65RGB and can confirm KC_SYSTEM_SLEEP & KC_SYSTEM_POWER are now working. Thanks for the update

zored, 2020-02-27:
@tzarc yes, sleep button works now. Thank youl

closed 2020-02-27 by zored