qmk/qmk_firmware #6204: Fresh checkout broken ergodox on Arch (w/ downgraded avr-gcc)

leshow, 2019-06-27:
I've downgraded avr-gcc to 8.3, however a fresh checkout of the repo will still fail to build the ergodox_infinity

```bash
> make git-submodule
> make ergodox_infinity:default MASTER=right
QMK Firmware 0.6.403
Making ergodox_infinity with keymap default

arm-none-eabi-gcc (Arch Repository) 9.1.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodox_infinity/keymaps/default/keymap.c                                      [OK]
Compiling: tmk_core/common/command.c                                                                [OK]
Compiling: lib/chibios/os/common/ports/ARMCMx/chcore_v7m.c                                         In file included from ./lib/chibios-contrib/os/common/ext/CMSIS/KINETIS/k20x7.h:134,
                 from ./lib/chibios-contrib/os/common/startup/ARMCMx/devices/K20x7/cmparams.h:70,
                 from ./lib/chibios/os/common/ports/ARMCMx/chcore.h:70,
                 from ./lib/chibios/os/rt/include/ch.h:81,
                 from lib/chibios/os/common/ports/ARMCMx/chcore_v7m.c:28:
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h: In function 'SVC_Handler':
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: error: listing the stack pointer register 'sp' in a clobber list is deprecated [-Werror=deprecated]
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:164:3: note: in expansion of macro '__ASM'
  164 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:164:3: note: in expansion of macro '__ASM'
  164 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h: In function '_port_irq_epilogue':
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: error: listing the stack pointer register 'sp' in a clobber list is deprecated [-Werror=deprecated]
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:164:3: note: in expansion of macro '__ASM'
  164 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/core_cm4.h:93:28: note: the value of the stack pointer after an 'asm' statement must be the same as it was before the statement
   93 |   #define __ASM            __asm                                      /*!< asm keyword for GNU Compiler */
      |                            ^~~~~
./lib/chibios/os/common/ext/CMSIS/include/cmsis_gcc.h:164:3: note: in expansion of macro '__ASM'
  164 |   __ASM volatile ("MSR psp, %0\n" : : "r" (topOfProcStack) : "sp");
      |   ^~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:372: .build/obj_ergodox_infinity/lib/chibios/os/common/ports/ARMCMx/chcore_v7m.o] Error 1
make: *** [Makefile:553: ergodox_infinity:default] Error 1
Make finished with errors
```

drashna, 2019-06-27:
You need to downgrade ARM GCC as well.

leshow, 2019-06-27:
I don't even have an arm-gcc installed, I built layouts as soon as 2 days ago on this machine. It has to have been a recent update.

leshow, 2019-06-27:
Oh I see: it's https://www.archlinux.org/packages/community/x86_64/arm-none-eabi-gcc/

closed 2019-06-27 by leshow