qmk/qmk_firmware #6206: Record Macro when holding Layer Tap-Toggle

jandamm, 2019-06-28:
## Describe the Bug

I've a Layer Tap-Toggle (Layer 4) button on my keyboard. In this Layer I've a shortcut to record/stop/run a macro.

1: When I tap the `TT`-Button to switch to the Layer 4 and then press `Record Macro` I get back to Layer 0 and am recording the macro. -> Everything is fine

2: When I hold `TT` and press `Record Macro` I'll stay in Layer 4 after releasing `TT`, but the macro thinks I'm in Layer 0:

As an example:
My Layout: https://configure.ergodox-ez.com/ergodox-ez/layouts/GG7Xa/latest/4

I'm in Layer 0
I hold `TT4` and start recording
I then press `a` `r`. -> Input: `a`, `move mouse left`
I quit Layer 4 `TO 0`
I hold `TT4` and press `s`. -> Input: `move mouse down`
I stop the macro.

Replay: `a` `r` `move cursor down`

## System Information

 - Keyboard:  
   Ergodox EZ
 - Operating system:
   macOS 10.14.5

drashna, 2019-06-28:
I'll have to look more into what you're talking about. 

But both the recording and playback assume that you're on layer 0 (or the default layer), with no other layers active.  In fact, both the functions for playback and recording clear the keyboard, and layer stack before starting (though playback stores the state, and restores it, when it's done).

mcp292, 2020-02-08:
My experience is similar. I hit record, do something, then hit stop. When I hit replay I get some funny behavior. Since the replay button is in the lower layer, the keystrokes of the macro translate to that layer or at least half of them until I release the toggle.

Linux
Planck EZ
Layout: https://configure.ergodox-ez.com/planck-ez/layouts/Oaj3d/latest/0

kosorin, 2020-10-14:
I have same problem. If I start recording macro while holding LWR and RSE keys, then layer is reset to 0 as you mentioned. But after releasing LWR and RSE layer is again changed back to layer with DM_REC1

I suspect this line: https://github.com/qmk/qmk_firmware/blob/b63b2106d818860038821f41596b92759068a3cb/tmk_core/common/action.c#L510
But I'm newbie and don't know how to fix it

zvecr, 2022-08-28:
This issue has been automatically closed because it has not had any recent activity.
If this issue is still valid, re-open the issue and let us know.

closed 2022-08-28 by zvecr

kosorin, 2022-08-29:
It's still an issue. And as I mentioned earlier one of the possible reason is using `layer_invert()` in `quantum/action.c`. https://github.com/qmk/qmk_firmware/blob/8b9f0a58b4a100bbae0fabeaa4ee29f2955d3dbf/quantum/action.c#L592-L603

1. be in layer `L0`
2. hold `TT(1)` and go to `L1`
3. start recording `DYN_REC_START1`
4. all layers are cleared
5. release `TT(1)` => layer is set back to `L1` (because of `layer_invert()`)

A simple fix would be replace `layer_invert` with the `layer_on`/`layer_off`, however this will slightly change `TT`'s behavior.
I proposed adding this as a new feature in #17261 (title and description isn't quite clear).