qmk/qmk_firmware #6208: Callback for key up event?

mvphilip, 2019-06-29:
Hello I am using ACTION_TAP_DANCE_FN_ADVANCED and I have set an action for the shift key.
Unfortunately per the doc if I have the shift held while typing another tap dance enabled key, the shift tap dance will be reset but with out an unregister event(?). I've been trying to work around that by storing shift state and then in process_record_user checking if the state has been cleared and !(get_mods() & MODS_SHIFT_MASK). But this is insufficient as  process_record_user does not seem to be triggered by the release of the shift key. 

So is there a callback for a key up event?

thank you.

-mp

drashna, 2019-07-03:
Just to make sure, I think you're looking for the "reset" function.  
https://docs.qmk.fm/#/feature_tap_dance?id=example-1-send-on-single-tap-on-double-tap

joukewitteveen, 2022-02-22:
This seemed to have been about "reset" and may well have been fixed since by the likes of #12471. I think this issue can be closed, but there are a few things worth noting for people stumbling on this issue.

First: The `on_each_tap_fn` callbacks is called after `process_record_user`, so the tap dance state will not have been updated yet and the tap dance logic will be subverted if you return `false`.

Second, note that "reset" is called in rather specific cases:
1. When the key is released after
    a. being held for longer than `TAPPING_TERM` milliseconds, or
    b. the tap dance is interrupted.
2. When the key was already released, but no other key was pressed for `TAPPING_TERM` milliseconds since the key was pressed.

Only in the first case does it correspond to a key release event. What tap dance doesn't offer is a callback for the *release* events of individual taps. The `on_each_tap_fn` is called on each *press* of the tap dance key.
The example added in #16394 shows how to act on release events of taps.

closed 2022-06-13 by drashna