qmk/qmk_firmware #6219: Minivan: K32 and K34 inverted on command and arrow_command layouts

francislan, 2019-07-01:
The command and arrow_command layouts has K32 and K34 inverted.

Keyboard:  TheVanKeyboards/Minivan

noroadsleft, 2019-07-01:
There's a possibility that you've triggered Bootmagic accidentally here as the Minivan has it enabled by default. Do you have the keymap that's installed on your Minivan uploaded to GitHub anywhere?

francislan, 2019-07-01:
Here's my [keymap].(https://github.com/francislan/qmk_firmware/blob/lan_keymap/keyboards/thevankeyboards/minivan/keymaps/lan/keymap.c)

francislan, 2019-07-01:
I have BootMagic disabled.

Furthermore, when I generate a JSON from http://qmk.thevankeyboards.com, I get:
"configKeymap":{"keys":["K00","K01","K02","K03","K04","K05","K06","K07","K08","K09","K0A","K0B","K10","K11","K12","K13","K14","K15","K16","K17","K18","K19","K1A","K1B","K20","K21","K22","K23","K24","K25","K26","K27","K28","K29","K2A","K2B","K30","K31",**"K32","K34"**,"K33","K37","K38","K39","K3A","K3B"],
"positions":[["K00","K01","K02","K03","K04","K05","K06","K07","K08","K09","K0A","K0B"],["K10","K11","K12","K13","K14","K15","K16","K17","K18","K19","K1A","K1B"],["K20","K21","K22","K23","K24","K25","K26","K27","K28","K29","K2A","K2B"],["K30","K31","K32","K33","K34","KC_NO","KC_NO","K37","K38","K39","K3A","K3B"]]}

The order is K32 -> K34 and not K34 -> K32 as it is right now.

drashna, 2019-07-03:
I believe this was fixed, recently.

closed 2019-07-03 by drashna

francislan, 2019-07-03:
The Pull request that fixes this issue is #6221. It hasn't been merged yet. I'm new to Gtihub and this repo, do we usually close the bug before the merge happens?

drashna, 2019-07-03:
Sorry, that was my bad.  I thought it was merged (eg, I thought that I had merged it). 

It's merged now, so hopefully everything is all good now. :)