qmk/qmk_firmware #6234: Can't get LGUI_T to work on Crkbd/rev1

jamztang, 2019-07-02:
It seemed like `LGUI_T(kc)` and `RGUI_T(kc)` on crkbd/rev1 doesn't register to the command key on macOS, the current workaround is to use `LGUI(KC_NO)`

May I know if that's a known issue?

fauxpark, 2019-07-03:
`LGUI_T(KC_A)` and `RGUI_T(KC_Z)` are working fine for me - not the same keyboard though. Which keycodes are you trying to use with these?

drashna, 2019-07-03:
It seems to work just fine for me, on my Corne.  

Could you post a link to your layout?

jamztang, 2019-07-04:
[image]

I tried using it with Backspace/Brackets/normal keys, it just doesn't seemed to work. Here's the link.

```
https://raw.githubusercontent.com/jamztang/qmk_firmware/master/james-2019-07-04.json
```

fauxpark, 2019-07-07:
@jamztang that JSON file just has a Backspace where the RGUI_T is in your image. I thought it might be a Configurator issue but if I change it back to the RGUI_T and download the JSON again it's there.

jamztang, 2019-07-08:
@fauxpark sorry I pointed to the wrong mapping file, link updated. The issue is that even if the `GUI_T` code is there, after flashing it doesn't work

drashna, 2019-07-08:
I have a corne keyboard, and I did test this out.   I can't reproduce the issue, at all.

So more than likely, this may be a bootmagic related issue. 

Add the `EEPROM Reset` keycode to your keymap, and hit that.  See if that fixes the issue for you.  

If not .... I'm not sure, as it works correctly for both modes, for me.

fauxpark, 2019-07-08:
Bootmagic seems to be disabled on the Corne :/

drashna, 2019-07-08:
Correct, but the settings are used regardless if bootmagic is enabled or not.  I really should say "magic".... but that may be more confusing. 

Specifically, it sounds like LGUI is explicitly disabled. `MAGIC_NO_GUI` specifically.

jamztang, 2019-07-08:
FYI, I am able to use the LGUI key, just not the LGUI_T. My two other friends with a couple of these keyboards has  the same issue.

haven’t tried resetting the EEPROM tho, will try out soon.

drashna, 2019-07-08:
Okay.  I just used the configurator to generate a keymap with both LGUI_T, and RGUI_T, and both worked just fine.

So.... I don't think that this is firmware related.   Are you using English US for the host system's layout?

drashna, 2019-07-08:
on an unrelated note, it may be worth splitting up the bootmagic doc page into "bootmagic" and "magic".

fauxpark, 2019-07-08:
Well, the feature formerly called "Magic" is now known as Command - the `MAGIC_` keycodes are the Bootmagic Keycodes: https://docs.qmk.fm/#/feature_bootmagic
And in fact it seems Command still uses the `MAGIC_` prefix for its config too.
Or are you suggesting the keycodes take over that name?

jamztang, 2019-07-08:
Good news, I think after I reset the EEPROM, it seemed to work now!! Thanks a lot!

drashna, 2019-07-09:
Bootmagic strikes again!  

Well, glad we got it resolved!

closed 2019-08-06 by jamztang