qmk/qmk_firmware #6257: Lily58 can't have an encoder on slave side

MysticMixles, 2019-07-05:
There is no way to use an encoder on the slave side of the Lily58 Pro. When the slave/right side is instead configured as the master side, and the index of the encoder is set to 0, it works as expected. When returned to normal, and the index set to either 0 or 1, it doesn't do anything. 

There are multiple mentions of this issue on the internet, and one user with custom code has fixed it, but it would likely be fixed if the board used split common code.

## System Information

 - Keyboard:  Lily58 Pro
   - Revision (if applicable):  
 - Operating system: Win10 64 bit
 - AVR GCC version: 5.4.0

 - ARM GCC version: 6.3.1

 - QMK Firmware version: 0.6.404

## Additional Context

yyc, 2019-07-06:
Hi @MysticMixles! As luck would have it, I just refactored the Lily58 code to use split_common.

There's currently a pull request open (#6260), and I'd appreciate it if you could review it (if able) or test it out to see if it's working for your use case.

viorels, 2020-04-19:
Hello, I've tried the lily58_refactor branch (latest commit 8647d94ea) of yyk fork but I can't get the encoder to work on either side (hardware side worked fine on one side with the stable code).

// rules.mk of lily58/keymap/default
ENCODER_ENABLE = yes

// config.h
#define NUMBER_OF_ENCODERS 1
#define ENCODERS_PAD_A { F5 }
#define ENCODERS_PAD_B { F4 }

Added encoder_update_user code from https://docs.qmk.fm/#/feature_encoders to keymap.c but NOT any code in <keyboard>.c

And nothing happens (expecting page up/down)... am I missing some config option? How can I debug this issue? Thanks

zvecr, 2022-08-28:
<https://github.com/qmk/qmk_firmware/blob/master/keyboards/lily58/glow_enc/config.h#L45-L48> for an example of a lily58 that supports encoders.

closed 2022-08-28 by zvecr