qmk/qmk_firmware #6277: RGB Light on Split Common doesn't properly sync

drashna, 2019-07-07:
## Describe the Bug

When using the `rgblight_sethsv_noepprom` functions for layer indication, the slave half rarely updates properly, getting stuck on the non-default layer color. 

This was introduced with #5020 and #5509 

## System Information

 - Keyboard:  Iris Rev2, Orthodox Rev3
 - Operating system: Windows 10, MacOS Mojave
 - AVR GCC version:  8.3.0
 - QMK Firmware version: 0.6.404

## Additional Context

Flash both sides with `make keebio/iris/rev2:drashna`, and use the layer keys.

razy69, 2019-08-02:
Hi,

I had the same issue with the Iris rev3.

config.h
``
#pragma once

#include QMK_KEYBOARD_CONFIG_H
#include "config_common.h"

/* USB Device descriptor parameter */
#define VENDOR_ID       0xCB10
#define PRODUCT_ID      0x1256
#define DEVICE_VER      0x0300
#define MANUFACTURER    Keebio
#define PRODUCT         Iris Keyboard
#define DESCRIPTION     Split 50 percent ergonomic keyboard

/* key matrix size */
// Rows are doubled-up
#define MATRIX_ROWS 10
#define MATRIX_COLS 6

// wiring of each half
#define MATRIX_ROW_PINS { D2, D3, D5, D7, D6 }
#define MATRIX_COL_PINS { F1, F4, F5, F6, D4, B4 }
#define SPLIT_HAND_PIN F0
#define QMK_ESC_OUTPUT F1
#define QMK_ESC_INPUT D2
#define QMK_LED B0
#define QMK_SPEAKER C6

#define ENCODERS_PAD_A { B5 }
#define ENCODERS_PAD_B { B7 }

/* COL2ROW or ROW2COL */
#define DIODE_DIRECTION COL2ROW

/* define if matrix has ghost */
//#define MATRIX_HAS_GHOST

/* number of backlight levels */
// #define BACKLIGHT_LEVELS 3

/* Set 0 if debouncing isn't needed */
#define DEBOUNCE 5

/* serial.c configuration for split keyboard */
#define SOFT_SERIAL_PIN D0

/* Mechanical locking support. Use KC_LCAP, KC_LNUM or KC_LSCR instead in keymap */
#define LOCKING_SUPPORT_ENABLE
/* Locking resynchronize hack */
#define LOCKING_RESYNC_ENABLE

#define BACKLIGHT_PIN B6
#define BACKLIGHT_LEVELS 5

/* ws2812 RGB LED */
#define RGB_DI_PIN F7
#define RGBLED_NUM 12    // Number of LEDs
#define RGBLED_SPLIT { 6, 6 }
#define RGBLIGHT_SLEEP
#define RGBLIGHT_EFFECT_BREATHING

#define DYNAMIC_KEYMAP_LAYER_COUNT 4

// EEPROM usage

// TODO: refactor with new user EEPROM code (coming soon)
#define EEPROM_MAGIC 0x451F
#define EEPROM_MAGIC_ADDR 32
// Bump this every time we change what we store
// This will automatically reset the EEPROM with defaults
// and avoid loading invalid data from the EEPROM
#define EEPROM_VERSION 0x08
#define EEPROM_VERSION_ADDR 34

// Dynamic keymap starts after EEPROM version
#define DYNAMIC_KEYMAP_EEPROM_ADDR 35
// Dynamic macro starts after dynamic keymaps (35+(4*10*6*2)) = (35+480)
#define DYNAMIC_KEYMAP_MACRO_EEPROM_ADDR 515
#define DYNAMIC_KEYMAP_MACRO_EEPROM_SIZE 509    // 1024-DYNAMIC_KEYMAP_MACRO_EEPROM_ADDR
#define DYNAMIC_KEYMAP_MACRO_COUNT 16

/* Use I2C or Serial, not both */
#define USE_SERIAL
//#define USE_I2C

/* Other */
//#define NO_USB_STARTUP_CHECK
#define LINK_TIME_OPTIMIZATION_ENABLE
#define PREVENT_STUCK_MODIFIERS
#define TAPPING_TERM 200
#define PERMISSIVE_HOLD
#define TAPPING_TOGGLE  3 // Double tab
#define AUTO_SHIFT_TIMEOUT 95
#define ONESHOT_TAP_TOGGLE 5  /* Tapping this number of times holds the key until tapped this number of times again. */
#define ONESHOT_TIMEOUT 5000  /* Time (in ms) before the one shot key is released */
``

keymap.c
``
#include QMK_KEYBOARD_H

extern keymap_config_t keymap_config;

#define _QWERTY 0
#define _RAISE 1

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
[_QWERTY] = LAYOUT(
KC_ESC , KC_1, KC_2, KC_3, KC_4, KC_5,                      KC_6, KC_7, KC_8   , KC_9  , KC_0   , KC_GRV ,
KC_TAB , KC_Q, KC_W, KC_E, KC_R, KC_T,                      KC_Y, KC_U, KC_I   , KC_O  , KC_P   , KC_BSLS,
KC_LCTL, KC_A, KC_S, KC_D, KC_F, KC_G,                      KC_H, KC_J, KC_K   , KC_L  , KC_SCLN, KC_QUOT,
KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_ENT,      KC_DEL, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT,
                    KC_LALT, KC_LGUI, KC_BSPC,      KC_SPC,TT(1), KC_RALT),

[_RAISE] = LAYOUT(
KC_TRNS,KC_F1  ,KC_F2,KC_F3,KC_F4,KC_F5,                                      KC_F6,KC_F7,KC_F8,KC_F9,KC_F10,KC_TRNS,
KC_TRNS,KC_F12 ,KC_NO,KC_UP,KC_NO,KC_POWER,                         KC__MUTE,KC_MRWD,KC_MPLY,KC_MFFD,KC_F11,KC_TRNS,
KC_TRNS,KC_F14 ,KC_LEFT,KC_DOWN,KC_RIGHT,KC_EQL,                       KC_MINS,KC_NO,KC_NO,KC__VOLUP,KC_F13,KC_TRNS,
KC_TRNS,KC_PGUP,KC_HOME,KC_END,KC_PGDN,KC_LBRC,KC_TRNS,     KC_INS,KC_RBRC,KC_NO,KC_NO,KC__VOLDOWN,KC_PSCR,KC_TRNS,
                            KC_TRNS,KC_TRNS,KC_TRNS,     KC_TRNS, KC_NO,KC_TRNS),
};

uint32_t layer_state_set_user(uint32_t state) {
    switch(biton32(state)) {
        case _RAISE:
            rgblight_sethsv_noeeprom(HSV_WHITE);
            break;
        default:
            rgblight_sethsv_noeeprom(HSV_RED);
            break;
    }
    return state;
}
``

When i quickly pushed the TT button, sometimes the master hand stuck on the WHITE layer color.
I switched the button to MO and now it works like a charm.

AngryEgret, 2019-11-07:
Can confirm the same on iris rev4:

keymap.c
```
#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
	[0] = LAYOUT(KC_GRV,KC_1,KC_2,KC_3,KC_4,KC_5,KC_6,KC_7,KC_8,KC_9,KC_0,KC_BSPC,KC_TAB,KC_Q,KC_W,KC_E,KC_R,KC_T,KC_Y,KC_U,KC_I,KC_O,KC_P,KC_BSLS,LCTL_T(KC_ESC),KC_A,KC_S,KC_D,KC_F,KC_G,KC_H,KC_J,KC_K,KC_L,KC_SCLN,KC_QUOT,KC_LSFT,KC_Z,KC_X,KC_C,KC_V,KC_B,MO(2),KC_MPLY,KC_N,KC_M,KC_COMM,KC_DOT,KC_SLSH,KC_SFTENT,KC_LGUI,LALT_T(KC_LBRC),KC_RBRC,KC_SPC,KC_MINS,KC_EQL),
	[1] = LAYOUT(KC_GRV,KC_1,KC_2,KC_3,KC_4,KC_5,KC_6,KC_7,KC_8,KC_9,KC_0,KC_BSPC,KC_TAB,KC_Q,KC_W,KC_E,KC_R,KC_T,KC_Y,KC_U,KC_I,KC_O,KC_P,KC_BSLS,KC_LSFT,KC_A,KC_S,KC_D,KC_F,KC_G,KC_H,KC_J,KC_K,KC_L,KC_SCLN,KC_QUOT,KC_LCTL,KC_Z,KC_X,KC_C,KC_V,KC_B,MO(2),KC_MPLY,KC_N,KC_M,KC_COMM,KC_DOT,KC_SLSH,KC_SFTENT,KC_LALT,KC_SPC,KC_SPC,KC_ESC,KC_MINS,KC_EQL),
	[2] = LAYOUT(KC_TRNS,KC_F1,KC_F2,KC_F3,KC_F4,KC_F5,KC_F6,KC_F7,KC_F8,KC_F9,KC_F10,KC_DEL,KC_TRNS,BL_TOGG,BL_DEC,BL_INC,KC_TRNS,KC_TRNS,TO(0),TO(1),KC_TRNS,KC_F11,KC_F12,KC_TRNS,KC_TRNS,RGB_TOG,RGB_SAI,RGB_SAD,RGB_HUI,RGB_HUD,KC_LEFT,KC_DOWN,KC_UP,KC_RGHT,KC_HOME,KC_PGUP,KC_TRNS,KC_TRNS,RGB_VAI,RGB_VAD,RGB_SPI,RGB_SPD,KC_TRNS,KC_MNXT,KC_TRNS,KC_TRNS,KC_TRNS,KC_TRNS,KC_END,KC_PGDN,KC_TRNS,KC_TRNS,KC_TRNS,KC_TRNS,KC_TRNS,KC_TRNS)
};

void encoder_update_user(uint8_t index, bool clockwise) {
	if (index == 1) {
        if (clockwise) {
            tap_code(KC_VOLU);
        } else {
            tap_code(KC_VOLD);
        }
    }
}

layer_state_t layer_state_set_user(layer_state_t state) {
    switch (get_highest_layer(state)) {
    case 0:
        rgblight_setrgb(RGB_RED);
        break;
    case 1:
        rgblight_setrgb(RGB_BLUE);
        break;
    case 2:
        rgblight_setrgb(RGB_GREEN);        
        break;
    }
  return state;
}```

GeorgePetri, 2020-04-06:
I have the same problem. But only on the nyquist rev3, I have another nyquist rev2 and it works as it should.
Does anyone have a fix?

drashna, 2020-05-25:
> I have the same problem. But only on the nyquist rev3, I have another nyquist rev2 and it works as it should.
> Does anyone have a fix?

I'm guessing the rev3 is split rgb, but the rev2 isn't (likely uses the "extra data" line).

tzarc, 2022-08-04:
Should be fixed.

closed 2022-08-04 by tzarc