qmk/qmk_firmware #6288: Add support for Massdrop x Zslane Mercury Rocketeer

jimelrod, 2019-07-09:
Per the [QMK Docs](https://docs.qmk.fm/#/newbs_building_firmware_configurator?id=selecting-your-keyboard), I'm requesting firmware to be available for the [Massdrop x Zslane Mercury Rocketeer](https://drop.com/buy/64755).

Cheers!

drashna, 2019-08-01:
Well, massdrop has it in their repo, but it doesn't look like it's done?  

https://github.com/Massdrop/qmk_firmware/tree/master/keyboards/massdrop/rocketeer

tofagerl, 2019-08-13:
I can't make heads or tails of the Massdrop rocketeer folder. It looks like it's really just support files for use with their web configurator, which I don't want to use since I already have a QMK keymap I use with other keyboards. There's not even a Makefile there.

I don't usually do C. Had this been a Java project I might have known what to do to put together the missing pieces, but as it is I'm useless, and so is this keyboard.

drashna, 2019-08-22:
Well, a PR is opened, so hopefully, it will get merged in soon.

closed 2022-08-04 by tzarc