qmk/qmk_firmware #6291: Support for Xtensa architecture

Skallwar, 2019-07-09:
## Architecture Support

- [ ] Support for Xtensa boards

## Description

Xtensa board such as Esp32 are really cheap boards but yet extremely powerful. For example, Esp32 board feature both WiFi and Bluetooth for less than [4$](https://fr.aliexpress.com/item/32801062068.html?src=google&albslr=201112132&albch=shopping&acnt=494-037-6276&isdl=y&slnk=&plac=&mtctp=&albbt=Google_7_shopping&aff_platform=google&aff_short_key=UneMJZVf&albagn=888888&albcp=1706977641&albag=66909914152&trgt=310549338888&crea=fr32801062068&netw=u&device=c&gclid=Cj0KCQjw9pDpBRCkARIsAOzRzit1zPcHnA8ytGp8o12G4Y07OrRmaO-ozV6kgqnWK6ZFNkYvQFP7W6saAk0AEALw_wcB&gclsrc=aw.ds) with a dual core SoC @ 240MHz.

I think this kind of beast are awesome and good option for custom/pro-grade keyboard, which should be paired with the equally awesome QMK firmware

docteurzoidberg, 2019-10-13:
yeah really should be considered, even for flash size of 4 to 16MB that would be great to add new features on QMK lcd screens, real pcm playback, integrated web server, and so on. possibilities are almost endless !

tzarc, 2022-08-03:
Someone would need to step up to implement this, and it's very much non-trivial.
Given that this has sat for nearly 3 years without progress, I'm going to go ahead and close it.

closed 2022-08-03 by tzarc