qmk/qmk_firmware #6334: Unknown Type name when compiling massdrop/alt:default

wildcat2083, 2019-07-15:
Compiling: keyboards/massdrop/alt/keymaps/default/keymap.c                                         keyboards/massdrop/alt/keymaps/default/keymap.c:123:1: error: unknown type name 'led_instruction_t'
 led_instruction_t led_instructions[] = {
 ^~~~~~~~~~~~~~~~~
keyboards/massdrop/alt/keymaps/default/keymap.c:125:5: error: braces around scalar initializer [-Werror]
     { .end = 1 }
     ^
keyboards/massdrop/alt/keymaps/default/keymap.c:125:5: note: (near initialization for 'led_instructions[0]')
keyboards/massdrop/alt/keymaps/default/keymap.c:125:7: error: field name not in record or union initializer
     { .end = 1 }
       ^
keyboards/massdrop/alt/keymaps/default/keymap.c:125:7: note: (near initialization for 'led_instructions[0]')
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [tmk_core/rules.mk:380: .build/obj_massdrop_alt_default/keyboards/massdrop/alt/keymaps/default/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:553: massdrop/alt:default] Error 1

wildcat2083, 2019-07-15:
I should mention i was using massdrop(s) fork of the qmk_firmware branch, it appears someone changed the branch 3 days ago now the default profile will not compile although the default_md will

drashna, 2019-07-15:
If the changes happened to a different repo, then you should open an issue on that repo, not here. 

As is, we don't know which repo you're referring to. Nor the change, in question.

wildcat2083, 2019-07-15:
I tried to but their isnt a report an issue tab there, the reason i reported it here was because the file in question appeared to be pulled from this fork instead of its own "copy" 

EDIT:: I figured it out, by removing the line of code for the led_instruct_t the compiler stopped complaining as it appears several repos all have removed the line

drashna, 2019-07-15:
Ah, okay.

Well, that looks to be a dev fork, so you may not want to use it.  It may be much safer to use the QMK branch, instead.

wildcat2083, 2019-07-15:
my thoughts as well :) 
quess ill close this off as the issue has been resolved

closed 2019-07-15 by wildcat2083