qmk/qmk_firmware #6354: RGB matrix toggle key no longer works

shieldsd, 2019-07-17:
## Describe the Bug

The keycode RGB_TOG no longer seems to work. I suspect this is related to the RGB re-factoring work carried out in May. Might also be related to enabling RGB_MATRIX_FRAMEBUFFER_EFFECTS. If I can track down the bug and fix it I'll submit a pull request.

## System Information

 - Keyboard:  Planck Light and Keyboardio Model 01
   - Revision (if applicable):  
 - Operating system: Fedora 29
 - AVR GCC version: 7.4

 - ARM GCC version: 7.4

 - QMK Firmware version: 0.6.429

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2019-07-19:
Neither seem to be an issue for me.

Have you tried resetting your eeprom?

Lenbok, 2019-07-30:
I saw something possibly related to this with my crkbd when I was alternately trying out rgblight vs rgb_matrix, at one point the RGB_TOG no longer functioned when I was on a rgb_matrix firmware but worked OK on rgblight ones. But at some point it came right and is working fine, so possibly it is an eeprom issue as drashna suggests.

drashna, 2019-08-01:
@shieldsd Were you still having this issue, after resetting the EEPROM?

shieldsd, 2019-08-06:
Re-setting the eeprom didn't seem to make any difference. I'm going to add an led off key rather than the toggle to see if that helps. Failing that, I'll bisect the git commits starting in May.

shieldsd, 2019-08-06:
Ok, some progress, re-setting the eeprom on my Model 01 has done the trick. I initialized through matrix_user_init rather than via a keycode. The same trick is not working for my Planck light though.

shieldsd, 2019-08-07:
Both working now. I removed the define for RGB_MATRIX_KEYPRESSES and re-set the eeprom. I think there's still a bug in there somewhere but I'll close this ticket for now.

closed 2019-08-07 by shieldsd