qmk/qmk_firmware #6359: Bluetooth functionality on AI03

JSchiltz19, 2019-07-18:
## Feature Request Type

Adding RN-42 support to AI03 number pad

## Description
I am trying to use a RN-42 ble module with to convert the ai03 number pad to wireless. I tried to add in the Bluetooth by connecting the TX and RX pins on the Arduino pro micro on the keyboard to the RX and TX pins on the RN-42 module respectively, and enable Bluetooth in the .mk file. This didn't work even after trying a couple different Bluetooth settings in the .mk file.

Maybe I'm just missing were to connect the Bluetooth too, but any help would be greatly appreciated.

Thank you
John Schiltz

drashna, 2019-08-01:
Which bluetooth module are you trying to add specifically?

JSchiltz19, 2019-08-01:
Specifically it is a HC-05 modual flashed with the RN-42 firmware. 

As an update... I created a library that allows the Bluetooth to work with any arduino by using  serial or the SoftwareSerial library by sending shorthand commands, but I had some issues integrating/compiling the libraries within the QMK firmware, so I'm still looking for native support.

drashna, 2019-08-01:
Well, in theory, we support RN-42 type modules.  It may be a good place to start. 

https://docs.qmk.fm/#/feature_bluetooth

However,  this only works for AVR/LUFA boards. No ARM support.

JSchiltz19, 2019-08-01:
The ATmega32U4 is AVR so that should work. Which pins do I connect to? Just the TX and RX on the arduino?

drashna, 2019-08-31:
I could be wrong, but looking at the RN42 support, it looks like it's broken in the repo. :(

I may be wrong .... but from what I can tell, some of the actual serial stuff is not present.

JSchiltz19, 2019-09-03:
That would make sense. I mean I do have a small arduino library that I made
that works for the RN-42, but I couldn't figure out how to integrate it
into the QMK firmware.

closed 2020-01-01 by stale[bot]