qmk/qmk_firmware #6380: Canoe backlight doesn't work

elmo-space, 2019-07-20:
The backlight on the Percent Studio Canoe doesn't work.

## Describe the Bug

The in switch backlight LEDs stay dark on the newest version of the Canoe PCB.
They work electrically (blinking when the board is in bootloader mode) but QMK doesn't seem to be able to turn them on anymore so they just stay dark.
The two RGB LEDs work as intended though.

## System Information

 - Keyboard:  
   - Percent Studio Canoe 60 (newest revision from R2)
 - Operating system: Windows 10 Pro 64 Bit
 - AVR GCC version: avr-gcc.exe (AVR_8_bit_GNU_Toolchain_3.6.1_1752) 5.4.0

 - ARM GCC version: arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

 - QMK Firmware version: preonic-1.0

 - Any keyboard related software installed? 
   - none

elmo-space, 2019-07-20:
Ok so turns out that in the rules.mk file for the board the setting BACKLIGHT_ENABLE is set to "no" per default which just disables the backlight.
Setting it to "yes" makes it work again.
Only thing that is a bit weird, is that the board has no brightness control for the backlight.
It just has 1 brightness level.

fauxpark, 2019-07-21:
The backlighting for this board appears to be split over several pins, which is... not ideal, but in fact supported (albeit without hardware PWM). Could you please apply and test this diff:

```diff
diff --git a/keyboards/canoe/canoe.c b/keyboards/canoe/canoe.c
index a7427e152..5d88c6cf3 100644
--- a/keyboards/canoe/canoe.c
+++ b/keyboards/canoe/canoe.c
@@ -16,32 +16,11 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */
 
 #include "canoe.h"
-#ifdef BACKLIGHT_ENABLE
-#include "backlight.h"
-#endif
 #ifdef RGBLIGHT_ENABLE
 #include "i2c.h"
 #include "rgblight.h"
 #endif
 
-#ifdef BACKLIGHT_ENABLE
-void backlight_set(uint8_t level) {
-	if (level == 0) {
-		// Turn out the lights
-		PORTD &= ~(1<<0 | 1<<1 | 1<<4 | 1<<6);
-	} else {
-		// Turn on the lights
-		PORTD |= (1<<0 | 1<<1 | 1<<4 | 1<<6);
-	}
-}
-
-void backlight_init_ports(void) {
-	DDRD |= (1<<0 | 1<<1 | 1<<4 | 1<<6);
-	PORTD &= ~(1<<0 | 1<<1 | 1<<4 | 1<<6);
-}
-
-#endif
-
 // for keyboard subdirectory level init functions
 // @Override
 void matrix_init_kb(void) {
diff --git a/keyboards/canoe/config.h b/keyboards/canoe/config.h
index cddb749dc..61046cccb 100644
--- a/keyboards/canoe/config.h
+++ b/keyboards/canoe/config.h
@@ -37,8 +37,9 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 #define DIODE_DIRECTION COL2ROW
 #define DEBOUNCE 5
 
-#define NO_BACKLIGHT_CLOCK
-#define BACKLIGHT_LEVELS 1
+#define BACKLIGHT_PINS { D0, D1, D4, D6 }
+#define BACKLIGHT_LEVELS 3
+
 #define RGBLIGHT_ANIMATIONS
 
 #define NO_UART 1
diff --git a/keyboards/canoe/rules.mk b/keyboards/canoe/rules.mk
index 3d7bc0215..fac85172e 100644
--- a/keyboards/canoe/rules.mk
+++ b/keyboards/canoe/rules.mk
@@ -36,7 +36,7 @@ MOUSEKEY_ENABLE = no
 EXTRAKEY_ENABLE = yes
 CONSOLE_ENABLE = yes
 COMMAND_ENABLE = yes
-BACKLIGHT_ENABLE = no
+BACKLIGHT_ENABLE = yes
 RGBLIGHT_ENABLE = yes
 RGBLIGHT_CUSTOM_DRIVER = yes
 
```

That removes the "custom" backlight functionality and reverts it back to the core QMK implementation, with proper brightness control and everything. There is even breathing support, but it's a little wonky at the dim end (so I didn't add it here).

elmo-space, 2019-07-22:
@fauxpark It seems to work, but it also makes the board very unstable.
Getting soft stuck keys and sudden disconnects with that.
It's probably the reason why this was deactivated for the ps2avr ported boards (it's the same for the Ares and the HB85).

closed 2019-07-22 by elmo-space

lborruto, 2020-02-13:
Hey, I'm sorry to comment here, but I cannot find any information online : how do I put my Canoe on bootloader mode and flash with QMK?

noroadsleft, 2020-02-17:
@lborruto,

According to [this Percent Studio link], hold Fn+Space for 3 seconds. However, I have seen no indication that this board runs QMK, or even is able to do so.