qmk/qmk_firmware #6394: Multiple `update_tri_layer_state` can't point to the same layer

OleMussmann, 2019-07-23:
## Describe the Bug

Chaining `update_tri_layer_state` works nicely, except when they point to the same target layer.

Works:
```
uint32_t layer_state_set_user(uint32_t state) {
  state = update_tri_layer_state(state, _LOWER, _RAISE, _ADJUST);
  state = update_tri_layer_state(state, _BLUE, _GREEN, _YELLOW);
  return state;
}
```

Doesn't work:
```
uint32_t layer_state_set_user(uint32_t state) {
  state = update_tri_layer_state(state, _LOWER, _RAISE, _ADJUST);
  state = update_tri_layer_state(state, _BLUE, _GREEN, _ADJUST);
  return state;
}
```

## System Information

 - Keyboard:  planck
   - Revision (if applicable):  rev6
 - Operating system: Ubuntu 18.04
 - AVR GCC version: avr-gcc (GCC) 5.4.0
 - ARM GCC version: arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
 - QMK Firmware version: 0.6.430
 - Any keyboard related software installed? No

drashna, 2019-08-01:
Yup, that's a known issue, and not something that really can be easily fixed. 

https://docs.qmk.fm/#/ref_functions?id=update_tri_layer_statestate-x-y-z

Specifically, the function checks to see if the first two layers are on. If so, it turns the third on. Else, it turns it off. 

You could add more checks, and handle this yourself, but .... you can't use the built in function.

Something like: 
```c
if ( ( layer_state_cmp(state, _LOWER) && layer_state_cmp(state, _RAISE ) ) || ( layer_state_cmp(state, _BLUE) && layer_state_cmp(state, _GREEN) ) ) {
    state | (1UL<<_ADJUST);
} else {
   state & ~(1UL<<_ADJUST);
}
```

drashna, 2019-08-22:
Is it okay if we close this?

OleMussmann, 2019-08-27:
I guess this issue is resolved. For completeness' sake, here's the code that worked for me, in case other people stumble upon this:

```
uint32_t layer_state_set_user(uint32_t state) {                                 
  if (                                                                          
      ( layer_state_cmp(state, _LOWER_1) && layer_state_cmp(state, _RAISE_1 ) ) ||
      ( layer_state_cmp(state, _LOWER_2) && layer_state_cmp(state, _RAISE_2 ) ) ||
      ( layer_state_cmp(state, _LOWER_3) && layer_state_cmp(state, _RAISE_3 ) ) ) {
    return state | (1UL<<_ADJUST);                                              
  } else {                                                                      
    return state & ~(1UL<<_ADJUST);                                             
  }                                                                             
}
```

I found that you gave advice to [others](https://www.reddit.com/r/olkb/comments/9wg1vl/help_qmk_update_tri_layer_state/) that have the same problem before. If this is indeed a know issue, maybe we could mention it in https://docs.qmk.fm/ to prevent future confusion?

Thanks for your help @drashna !

jweickm, 2021-08-22:
I tried your code and it works really well. Thanks for sharing this! :)

drashna, 2024-10-13:
Closing, is this is resolved

closed 2024-10-13 by drashna