qmk/qmk_firmware #6403: Can't flash my preonic rev3

missylo, 2019-07-24:
I am able to successfully make a bin file with my keymap using the command: make preonic/rev3:Missylo

However, I am getting this error when i try to  run: make preonic/rev3:Missylo:dfu-util

Deducing device DFU version from functional descriptor length
dfu-util: No DFU capable USB device available
make[1]: *** [dfu-util] Error 74
make: *** [preonic/rev3:default:dfu-util] Error 1
Make finished with errors

I've also tried to do this with the default keymap, but am getting this error: 

1s-iMac:qmk_firmware a1$ make preonic/rev3:default:dfu-util
QMK Firmware 0.6.436
Making preonic/rev3 with keymap default and target dfu-util

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  49284	      0	  49284	   c084	.build/preonic_rev3_default.hex

Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/preonic_rev3_default.elf                                                            [OK]
Creating binary load file for flashing: .build/preonic_rev3_default.bin                             [OK]
Creating load file for flashing: .build/preonic_rev3_default.hex                                    [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  49284	      0	  49284	   c084	.build/preonic_rev3_default.hex

Copying preonic_rev3_default.bin to qmk_firmware folder                                             [OK]
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length
dfu-util: No DFU capable USB device available
make[1]: *** [dfu-util] Error 74
make: *** [preonic/rev3:default:dfu-util] Error 1
Make finished with errors

Any help would be appreciated! thank you

missylo, 2019-07-24:
when i try to run a check-size, i get this: 
1s-iMac:qmk_firmware a1$ make preonic/rev3:default:check-size
QMK Firmware 0.6.436
Making preonic/rev3 with keymap default and target check-size

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  49284	      0	  49284	   c084	.build/preonic_rev3_default.hex

Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/preonic_rev3_default.elf                                                            [OK]
Creating binary load file for flashing: .build/preonic_rev3_default.bin                             [OK]
Creating load file for flashing: .build/preonic_rev3_default.hex                                    [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  49284	      0	  49284	   c084	.build/preonic_rev3_default.hex

Copying preonic_rev3_default.bin to qmk_firmware folder                                             [OK]
(Firmware size check does not yet support cortex-m4 microprocessors; skipping.)

fauxpark, 2019-07-24:
Is the board in bootloader mode when you try to flash?

If it is, you probably need to install the WinUSB driver with Zadig: https://zadig.akeo.ie/

missylo, 2019-07-24:
Hi thank you for replying! 

I believe it’s in bootloader mode - hitting the reset button puts in in bootloader mode, right? 

Ive tried it via QMK toolbox as well. When I hit the reset button, it says STM32 is connected. I’m assuming this is the bootloader mode? 

Will Zadig work with Mac OS? I read that it’s for windows (driver issue?). 

Thank you for replying!

fauxpark, 2019-07-24:
Sorry, I didn't see you were on a Mac - in that case yes, Zadig is not needed as macOS should have the necessary drivers to talk to it.

The reset button does put it into bootloader mode, and if the Toolbox picks it up that's a good sign. Does trying to flash with it produce the same error message?

missylo, 2019-07-24:
Thanks again for responding!

When I try to flash it on QMK toolbox, I get this error:

\*\*\* OLKB - Preonic disconnected -- 0xFEED:0x6061

\*\*\* STM32 device connected

\*\*\* Attempting to flash, please don't remove device

\>>> dfu-util -a 0 -d 0482:df11 -s 0x8000000:leave -D /Users/a1/Desktop/k

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.

Copyright 2010-2016 Tormod Volden and Stefan Schmidt

This program is Free Software and has ABSOLUTELY NO WARRANTY

Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length

Opening DFU capable USB device...

ID 0483:df11

Run-time device DFU version 011a

Claiming USB DFU Interface...

dfu-util: Cannot claim interface

I get this error Every. single. time..... 
I’ll unplug it, it will say STM32 device unconnected. When I plug it back in, QMK Toolbox recognises it as OLKB Preonic Rev 3. And it says keyboard ready.

fauxpark, 2019-07-24:
Can you run `dfu-util -l` with the board in bootloader mode and paste what it says here?

missylo, 2019-07-25:
It says: 
1s-iMac:qmk_firmware a1$ dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length
Found Runtime: [05ac:828b] ver=0137, devnum=6, cfg=1, intf=3, path="29-1.8.1.3", alt=0, name="UNKNOWN", serial="UNKNOWN"
Found DFU: [0483:df11] ver=2200, devnum=7, cfg=1, intf=0, path="20-4.3", alt=1, name="@Option Bytes  /0x1FFFF800/01*016 e", serial="207737542039"
Found DFU: [0483:df11] ver=2200, devnum=7, cfg=1, intf=0, path="20-4.3", alt=0, name="@Internal Flash  /0x08000000/128*0002Kg", serial="207737542039"

missylo, 2019-07-25:
I asked on reddit, and a user suggested I try Sudo before the command.. and it worked!!!

-iMac:qmk_firmware a1$ sudo make preonic/rev3:default:dfu-util
Password:
QMK Firmware 0.6.436
Making preonic/rev3 with keymap default and target dfu-util

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  49284	      0	  49284	   c084	.build/preonic_rev3_default.hex

Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/preonic_rev3_default.elf                                                            [OK]
Creating binary load file for flashing: .build/preonic_rev3_default.bin                             [OK]
Creating load file for flashing: .build/preonic_rev3_default.hex                                    [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  49284	      0	  49284	   c084	.build/preonic_rev3_default.hex

Copying preonic_rev3_default.bin to qmk_firmware folder                                             [OK]
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 49288
Download	[=========================] 100%        49288 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state

missylo, 2019-07-25:
I can confirm that the sudo command works for me. I am able to flash my own keymap using the sudo command. Thank you!!!!

drashna, 2019-07-25:
If `sudo` worked, then it's a permission/access issue.

For long term, it may be better to not use sudo, but that means that you'll have to add udev rules:
https://docs.qmk.fm/#/faq_build?id=linux-udev-rules

fauxpark, 2019-07-25:
No udev on macOS, this is something else.

missylo, 2019-07-25:
Thank you so much for the reddit comment, @drashna!! at leat i have the default map and ADJUST function back.

I'm not sure how to proceed moving forward. I'm a complete novice. First time building anything via terminal (well, technically, i didn't do anything, just typed in a couple of commands that docs.qmk.fm told me to). 

@fauxpark is there any other info i can give you to help figure out what this issue is?

closed 2019-12-21 by stale[bot]