qmk/qmk_firmware #6407: Macro strings

capaldo, 2019-07-24:
Is it possible to assign a sequence of keys to trigger a macros. For example `\bc` to send the string "because"?

noroadsleft, 2019-07-24:
I believe [Leader Key](https://docs.qmk.fm/#/feature_leader_key) is the feature you want.

danielo515, 2019-07-26:
Leader key as @noroadsleft suggested is exactly what you want. I was about to open an issue to request an improvement for it.
I think you are already familiar with the concept of leader key since your example starts with  `\`, which is usually the leader key on vi.

capaldo, 2019-07-26:
> Leader key as @noroadsleft suggested is exactly what you want. I was about to open an issue to request an improvement for it.
> I think you are already familiar with the concept of leader key since your example starts with `\`, which is usually the leader key on vi.

It does. Thank you for both your help. Much appreciated.

danielo515, 2019-07-27:
So maybe close the issue?

capaldo, 2019-07-30:
Sorry about that forgot to close it.

closed 2019-07-30 by capaldo