qmk/qmk_firmware #6409: Implement layer mod just when held

danielo515, 2019-07-25:
Is it possible to switch to a layer with a modifier active just when held?
I mean, lacher switch with mod when held and keycode when tapped. Exactly like `LT` does.
If not, it will be cool to implement it.

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

PierreKircher, 2019-08-01:
tapdance is what you looking at 
https://beta.docs.qmk.fm/features/feature_tap_dance

danielo515, 2019-08-01:
Well, not exactly.
I want to act as a layer switcher with a modifier active when held, and as a normal key when tapped.
Tap-dance has a different objective,and a lot of boilerplate that I really hate.

PierreKircher, 2019-08-01:
its not about what you like 

you can do the same in tapdance which you can in a macro .. aka you can send a keycode - activate a layer on some action 
and other on other - you need to implement the custom tapdance for sure 

on tap - 

register_code(KC_RSFT);
ungister_code(KC_RSFT);

and for layer switch -

layer_on(_LOWER);
layer_off(_LOWER);

username11236, 2019-12-04:
This should do it:

typedef struct {
  bool is_press_action;
  int state;
} tap;

enum {
  SINGLE_TAP = 1,
  SINGLE_HOLD = 2
};

enum {
  NONE = 0,
  QUOT_LAYR,
};

int cur_dance (qk_tap_dance_state_t *state);

void ql_finished (qk_tap_dance_state_t *state, void *user_data);
void ql_reset (qk_tap_dance_state_t *state, void *user_data);

enum keyboard_layers {
  _BASE = 0,        // Base Layer
  _NAVC,    // Function Layer
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
[_BASE] = KEYMAP(KC...   , TD(QUOT_LAYR), .....
		),

[_NAVC]	= KEYMAP(KC...
    ),
};

int cur_dance (qk_tap_dance_state_t *state) {
  if (state->count == 1) {
    if (!state->pressed) {
      return SINGLE_TAP;
    } else {
      return SINGLE_HOLD;
    }
  }
  else return 8;
}

static tap ql_tap_state = {
  .is_press_action = true,
  .state = 0
};

void ql_finished (qk_tap_dance_state_t *state, void *user_data) {
  ql_tap_state.state = cur_dance(state);
  switch (ql_tap_state.state) {
    case SINGLE_TAP: 
      register_code(KC_A);
      break;
    case SINGLE_HOLD: 
      if (!layer_state_is(_NAVC)) {layer_on(_NAVC);register_code(KC_LSFT);}
      break;
  }
}

void ql_reset (qk_tap_dance_state_t *state, void *user_data) {
  if (ql_tap_state.state==SINGLE_HOLD) {
    if(layer_state_is(_NAVC)) {layer_off(_NAVC);}
    unregister_code(KC_LSFT);
  }
  if (ql_tap_state.state==SINGLE_TAP) {
    unregister_code(KC_A);
  }
  ql_tap_state.state = 0;
}

qk_tap_dance_action_t tap_dance_actions[] = {
  [QUOT_LAYR] = ACTION_TAP_DANCE_FN_ADVANCED_TIME(NULL, ql_finished, ql_reset, 175),
};

ridingqwerty, 2019-12-06:
Hey @danielo515 

I also feel like tapdance is a bit overkill for this functionality.  It's doable within `process_record_user` -- here's the case I use activates my `_NUMBER` layer with `KC_RGUI` active when held, but types `KC_QUOT` on a tap when the enumerated `RG_QUOT`keycode is used, which uses much less overhead than tapdance would:

```
    case RG_QUOT:
      if (record->event.pressed) {
        user_key_timer = timer_read();
        layer_on(_NUMBER);
        register_mods(MOD_BIT(KC_RGUI));
      } else {
        unregister_mods(MOD_BIT(KC_RGUI));
        layer_off(_NUMBER);
        if (timer_elapsed(user_key_timer) < TAPPING_TERM) {
          register_code(KC_QUOT);
          unregister_code(KC_QUOT);
        }
      }
      return false; break;

```

closed 2020-04-04 by stale[bot]