qmk/qmk_firmware #6428: qmk compile-json crashes with TypeError

focusaurus, 2019-07-28:
## Describe the Bug

Trying to run `qmk-compile-json path/to/my/layout.json` crashes with a `TypeError`. Looks like passing an unexpected argument to `subprocess.run`. I'm trying to run with a python3 virtualenv.

TraceBack:

```
./bin/qmk-compile-json ~/projects/ergodox/config.qmk.fm/2019-01-28-qwe-navpunc-num.json
Traceback (most recent call last):
  File "./bin/qmk-compile-json", line 41, in <module>
    result = subprocess.run(command, text=True, capture_output=True)
  File "/usr/lib/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
TypeError: __init__() got an unexpected keyword argument 'text'
```

## System Information

 - Keyboard:  infinity ergodox
 - Operating system: Linux Mint 19.1 Cinnamon
 - AVR GCC version: 

 - ARM GCC version: avr-gcc (GCC) 5.4.0

 - QMK Firmware version: 0.6.439

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I didn't initially have `pip3` available. I tried running `util/linux_install.sh` without any `pip3`, after activating a python3 virtualenv where `pip3` was early in the PATH, and after manually installing the `python3-pip` debian package.

I think you may need to add `python3-pip` to your required package list for debian-based distros, but even after installing that I get the `TypeError`.

focusaurus, 2019-07-28:
Note I was able to work around this and get the command to work by commenting out some stuff as per this diff:

```diff
diff --git a/bin/qmk b/bin/qmk
index c34365bed..40af28d13 100755
--- a/bin/qmk
+++ b/bin/qmk
@@ -37,12 +37,12 @@ with open('requirements.txt', 'r') as fd:
             exit(255)
 
 # Figure out our version
-command = ['git', 'describe', '--abbrev=6', '--dirty', '--always', '--tags']
-result = subprocess.run(command, text=True, capture_output=True)
+#command = ['git', 'describe', '--abbrev=6', '--dirty', '--always', '--tags']
+#result = subprocess.run(command)
 
-if result.returncode == 0:
-    os.environ['QMK_VERSION'] = 'QMK ' + result.stdout.strip()
-else:
+#if result.returncode == 0:
+#    os.environ['QMK_VERSION'] = 'QMK ' + result.stdout.strip()
+#else:
 os.environ['QMK_VERSION'] = 'QMK ' + strftime('%Y-%m-%d-%H:%M:%S')
 
 # Setup the CLI

```

drashna, 2019-08-01:
@skullydazed

skullydazed, 2019-08-07:
Thanks for your report here. Sorry for the delay looking at it I've been traveling and am now working through my backlog.

It looks like I accidentally used a feature that was added in python 3.7. I'll generate a PR to fix that.

skullydazed, 2019-08-07:
Can you `git pull` and give it a try again? It should be working for you now.

closed 2019-08-07 by skullydazed

focusaurus, 2019-08-09:
Looks like it's still using `capture_output` which is a new kwarg in 3.7.

```
./bin/qmk-compile-json ~/projects/ergodox/config.qmk.fm/2019-01-28-qwe-navpunc-num.json
Traceback (most recent call last):
  File "./bin/qmk-compile-json", line 41, in <module>
    result = subprocess.run(command, universal_newlines=True, capture_output=True)
  File "/usr/lib/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
TypeError: __init__() got an unexpected keyword argument 'capture_output'
```

focusaurus, 2019-08-09:
I hacked the shebang to run `python3.7` but I'm stuck in a loop where I get this message:

```
Your QMK build environment is not fully setup!

Please run `./util/qmk_install.sh` to setup QMK.
```

So I run `./util/qmj_install.sh` but it keeps telling me it's not setup still.

skullydazed, 2019-08-25:
Sorry for the delay getting back to this. #6592 should fix you up now. I've also opened #6588 which should help us catch this sort of thing in the future.

focusaurus, 2019-08-25:
Awesome, thanks! I can confirm it works without modification for me now.

PR #6506 (2019-08-07): Use the older `universal_newlines` name instead of `text`

## Description

## Types of Changes

- [x] Core
- [x] Bugfix

## Issues Fixed or Closed by This PR

* fixes #6428

## Checklist

- [x] My code follows the code style of this project.
- [x] I have read the [**CONTRIBUTING** document](https://docs.qmk.fm/#/contributing).
- [x] I have tested the changes and verified that they work and don't break anything (as well as I can manage).