qmk/qmk_firmware #6437: KC_GRV & KC_NUBS inverted on macOS

MrJuliuss, 2019-07-29:
# KC_GRV & KC_NUBS inverted on macOS

## Describe the Bug

Hi !
I have a problem with `KC_NUBS` & `KC_GRV`, on my layout. They seem to be inverted on macOS. I test the pcb on Windows 10, and all is OK. I will test on Ubuntu tomorrow.

Regards,

Julien

## System Information

 - Keyboard:  dz60
   - Revision:  Rev3
 - Operating system: macOS 10.14.6
 - AVR GCC version: 8.3.0
 - ARM GCC version: 7.3.1 20180622
 - QMK Firmware version: 0.6.439
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

fauxpark, 2019-07-29:
You probably need to set the correct keyboard type in macOS. In the Keyboard preferences on the first tab should be a button labelled "Change Keyboard Type...". Click that and follow the instructions, and it should fix the incorrect layout.

MrJuliuss, 2019-07-30:
Hi, indeed my keyboard was recognized as an ansi layout. It's now ISO and it works :)

Thanks !

closed 2019-07-30 by MrJuliuss

metawops, 2021-08-14:
I'm having the same issue (with another keyboard but QMK firmware)  but in macOS Big Sur I can't find any "Change Keyboard Type ..." button in the Keyboard System Preferences ... Any hint/help?