qmk/qmk_firmware #6471: Enter Key and \| key seem to be swapped in the matrix. Melody96

rockydbull, 2019-08-02:
When using the qmk config website I have to move the enter key to the |\ position and the |\ key to the enter key position for them to properly work on their correct keys IRL according to the ansi standard. This is with the melody96 hotswap ansi board with the 1u bottom row.

noroadsleft, 2019-08-02:
@zunger-humu,

You added the `LAYOUT_hotswap` macro to the melody96. Does [your keymap](https://github.com/qmk/qmk_firmware/blob/d68d5104731fc70d63c0769e3093bedfbaf7b694/keyboards/melody96/keymaps/zunger/keymap.c) work as intended when compiled from a current QMK master, and do you use ANSI Enter or ISO Enter? I looked at the macros and the `LAYOUT_hotswap` macro doesn't assign `KC_BSLS` on your keymap to the same place in the matrix as the other two macros would do.

Would appreciate your insight on this.

rockydbull, 2019-08-03:
@noroadsleft I think you may have replied to the wrong user (I also might be misreading this). I didnt make a layout. I am having the flipped |\ and enter key issue on the default layout with QMK. I am using Ansi enter for reference.

zunger-humu, 2019-08-03:
Whoops -- sorry I missed your message earlier. I'm using it with ANSI
Enter. I haven't compiled it in about three months, so I'm not sure if it
works at current head, but there hasn't been anything changed that I know
of which should affect this.

The assignment of KC_BSLS to a different place was deliberate: when I
applied the other two keymaps to a hotswap melody96, that key was
mismapped. (As were a few others; that's why I added this keymap in the
first place)

noroadsleft, 2019-08-03:
@zunger-humu Thanks for responding.

@rockydbull I requested zunger-humu's input because he's the user who submitted the Melody96 `LAYOUT_hotswap` macro to QMK, and I wanted to make sure that it was working for him. Sometimes a user will submit code and then discover it doesn't work correctly after their submission has been merged, so they fix it locally but then never submit those fixes back to us. zunger-humu has no history of doing this, but I wanted to make sure as I don't own the board in question.

Actually, looking over your initial post, @rockydbull, are you just flashing the default keymap that loads when you select `melody96` as your keyboard in QMK Configurator?

zunger-humu, 2019-08-03:
I'm typing on said keyboard right now, in fact. :)

@noroadleft, that would be my next question, too; the default melody96 keymap definitely didn't work for me on the hotswap. My keymap may be a bit on the intense side (the ability to type in Greek letters and math symbols and such) but you should definitely be able to use it, or build your own with the macros added.

(Which reminds me that I had some improvements to that keymap in mind, I should do those)

PR #22113 (2023-09-24): YMDK Melody96 Matrix Touch-Up
## Description

This PR fixes a long-standing bug regarding the ANSI Backslash and Enter keys. I meant to chase this down long ago.

- Correct layout/matrix diagram / Correct matrix data
  Swaps the matrix designations for the ANSI Backslash and ANSI Enter keys on Melody96 soldered PCBs, which has been broken since #16455.
- Implement `LAYOUT_hotswap` directly
  This is a temporary fix; I have a branch in progress to break up the support for Soldered and Hotswap PCBs into their own directories (the Hotswap PCB is multi-layout).
- Specify primary support for Solder [... cut]
closed 2023-09-24 by noroadsleft