qmk/qmk_firmware #6474: Cannot Flash Massdrop Alt

GrinninBarrett, 2019-08-02:
Hello,

I'm sorry if this is an annoying nonsense support request. People who understand this stuff probably do it so easily. But I have flashed keyboards before with no issues, so I'm not sure what the problem is here. 

I have had the Massdrop Alt for over a month now but finally decided to configure some different RGB options, specifically just adding different solid colors. I assumed this would be really straightforward. I used the configurator on Massdrop and added the new colors (which I assumed would simply appear in the cycle of stock RGB modes cycled with FN + A/D). I'm actually still not sure if that's how it will work, but I did that, saved, compiled, and downloaded that keymap. 

I followed all the instructions as carefully as I could. I have the latest version of Toolbox (or I'm fairly certain I do). I have flashed keyboards before, so my computer has the qmk_firmware directory and all of that stuff. I saved the executable and applet-*.bin files to the qmk_firmware folder. Using Terminal, I navigated to the qmk_firmware directory. Everything seemed normal up to this point. But when I tried to run the mdloader_mac command as listed at https://github.com/Massdrop/mdloader, I could not get anything to happen. It always says "command not found". I also tried simply opening the downloaded keymap into Toolbox, putting the keyboard into DFU mode, and flashing, and it just gives some errors and says "no device detected" and things like that. 

It seems like this should not be so complicated. What am I doing wrong? How can I go about making such a simple change and flashing it to the keyboard?

Thank you for your help.
Tucker

GrinninBarrett, 2019-08-02:
I apologize for wasting your time! I have figured it out and no longer require help. I realized what I was doing wrong, flash-wise, and also learned that using the configurator I can't simply add colors to the existing RGB color cycles. I've made the colors what I like and customized some options on different layers, and have successfully flashed the keyboard. Feel free to close this, and sorry for the confusion.

noroadsleft, 2019-08-02:
Glad you got it sorted. Closing.

closed 2019-08-02 by noroadsleft