qmk/qmk_firmware #6495: MSYS2 unable to install USB drivers

MrHarles, 2019-08-06:
I'm not even sure this is the correct forum for this question but I don't really know where else.. 

I was getting started using MSYS2 for the first time -- downloading drivers and installing the whatnots when I became stuck at the USB driver install prompt

```
Which USB drivers do you want to install?
(A)all - All supported drivers will be installed
(C)onnected - Only drivers for connected keyboards (in bootloader/flashing mode)
              will be installed
(F)force - Like all, but will also override existing drivers for connected
           keyboards
(N)one - No drivers will be installed,
         flashing your keyboard will most likely not work
(A/C/F/N)? A
~/qmk_utils /c/users/mrtor/downloads/qmk_firmware/qmk_firmware/util /c/users/mrtor/downloads/qmk_firmware/qmk_firmware

Microsoft Windows [Version 10.0.17763.615]
(c) 2018 Microsoft Corporation. All rights reserved.

C:\msys64\home\MrTor\qmk_utils>
```

I am super new to both QMK and MSYS, but am super eager to learn and progress.

Any help would be greatly appreciated!

MrHarles, 2019-08-06:
type exit

closed 2019-08-06 by MrHarles

fauxpark, 2019-08-06:
Typing `exit` is not the solution here, although the script continues, the driver installer will not be run. You will have to install the drivers through Zadig instead.

There is a bug in the way that the script is trying to run the executable - hopefully fixed in #6488.