qmk/qmk_firmware #6518: SEND_STRING on LEADER repeats first character

cquiroz, 2019-08-10:
## Describe the Bug

I recompiled my ergodox infinity keymap and now all the LEADER keys macros that use SEND_STRING are broken. If for example I have

```
    SEQ_TWO_KEYS (KC_Q, KC_X) {
      SEND_STRING("cquiroz");
    }

```

The keyboard used to send the "cquiroz" keys and now it keeps sending the "c" key and I need to restart the keyboard unplugging it. I verified that the keyboard actually keeps sending that key and it is not at the os software level.

Also if instead of SEND_STRING I send keys one by one it works well

I've traced back my changes and the big change was an upgrade of the tool chain to avr-gcc 8 but going back to avr-gcc 7 doesn't solve it

Any ideas?

## System Information

 - Keyboard:  
   ergodox_infinity
   - Revision (if applicable):  
 - Operating system:
OSX
 - AVR GCC version: 

avr-gcc (GCC) 8.3.0
 - ARM GCC version: 

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
 - QMK Firmware version:

0.6.447
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

cquiroz, 2019-08-10:
An extra data point. The code still works fine for my other pro-micro based keyboards

drashna, 2019-08-22:
Could you post a link to the boards in question, and your keymap? 

Also, have you tried adding `wait_ms(10)` before the send string to see if that helps?

cato976, 2019-10-03:
@drashna I have tried add a wait_ms(10) but issue still persist.  In addition I see the same behavior with the LT() "Toggling Layers" method.  LT(_MDIA_MOUSE, KC_O) use to work without issue.  now when I type "o",  I get the first "o" then it continues to repeat and I have to unplug the keyboard.

cato976, 2019-10-17:
After merging my fork from upstream LT(_MDIA_MOUSE, KC_O) now works as expected.

closed 2020-02-14 by stale[bot]