qmk/qmk_firmware #6519: Need help adding Zenith Z150 as sub-keyboard to xt_usb

rjmigliori, 2019-08-10:
I have a Zenith Z150 which I've successfully converted to USB using QMK. I accomplished this by duplicating the xt_usb directory in converter and renaming and modifying some files. I can provide details of how I did this on request, but the directory structure looks like this 

```
ZenithZ150/
├── config.h
├── info.json
├── keymaps
│   └── default
│       ├── config.h
│       └── keymap.c
├── led.c
├── matrix.c
├── README.md
├── rules.mk
├── ZenithZ150.c
└── ZenithZ150.h
```

and I can compile with 
`make converter/ZenithZ150:default`

However, since the Zenith is just an xt keyboard, I feel it more appropriately belongs inside the xt_usb directory, so that the directory structure looks something like this 

```
xt_usb/
├── config.h
├── info.json
├── keymaps
│   └── default
│       ├── config.h
│       └── keymap.c
├── led.c
├── matrix.c
├── README.md
├── rules.mk
├── xt_usb.c
├── xt_usb.h
└── ZenithZ150
    ├── file_1.ext
    ├── file_2.ext
    ├── file_n.ext
    └── keymaps
        └── default
            ├── config.h
            └── keymap.c
```
and is compiled with a command like 

`make converter/xt_usb/ZenithZ150:default`

Compare with the sun_usb directory, which operates exactly as I'm talking about 

sun_usb/
├── command_extra.c
├── config.h
├── led.c
├── matrix.c
├── readme.md
├── rules.mk
├── type3
│   ├── info.json
│   ├── keymaps
│   │   └── default
│   │       └── keymap.c
│   ├── rules.mk
│   └── type3.h
├── type4
│   ├── info.json
│   ├── keymaps
│   │   ├── default
│   │   │   └── keymap.c
│   │   └── sigma
│   │       ├── keymap.c
│   │       ├── readme.md
│   │       └── rules.mk
│   ├── rules.mk
│   └── type4.h
└── type5
    ├── info.json
    ├── keymaps
    │   ├── default
    │   │   └── keymap.c
    │   └── sigma
    │       ├── keymap.c
    │       ├── readme.md
    │       └── rules.mk
    ├── rules.mk
    └── type5.h

Which I have successfully added the type4, exactly as the way I'm trying to add the Zenith to xt_usb. 

Happy to answer questions or clarify anything I've said! Just trying to do this right. Thanks!

noroadsleft, 2019-11-21:
Hey, sorry your question fell through the cracks.

If you push your existing code to a GitHub fork, it would be much easier to help you work this out.

rjmigliori, 2019-11-21:
Okay I'll get around to it sometime in the next 30 days. Thanks for getting
back to me.

closed 2020-03-20 by stale[bot]