qmk/qmk_firmware #6530: RGB per-key support

icyhat, 2019-08-13:
Hi there guys,

Recently I made a hand-wired keyboard followed the instructions on kbfirmware.com, I wonder is there any RGB lights per-key support on QMK keyboards solution so I can custom light on each key.

The official doc shows RGB underglow, can they be put under each key?

zvecr, 2019-08-13:
You want https://docs.qmk.fm/#/feature_rgb_matrix, but thats unlikely to be in the kbfirmware code, as its based off old code (quoted to be over 2 years at this point).

icyhat, 2019-08-14:
Thanks your reply, kind man.

WS2812 seems like backlighting light, which usually support 16 LEDs at most on some QMK keyboards.

IS31FL3731 seems like the solution, I only found an official 30-pages doc with complicated terminologies of the driver, where can I find a simple user-oriented tutorial to buy needed LEDs and how to connect the IS31FL3731 and Atmega32U4 MCU, will docs of QMK contain these content?

zvecr, 2019-08-14:
There is WS2812 support for RGB matrix, the driver config is on that same page <https://docs.qmk.fm/#/feature_rgb_matrix?id=ws2812-avr-only>. WS2812 are not limited to 16, but generally to the power limits of USB. For instance, some of the rgbkb boards have over 100.

icyhat, 2019-08-14:
oh, that's what I want, per key LED.

I saw the configs of QMK files about WS2812, but what should I do before setting these config, how to connect physical WS2812 to pro micro.

Also, due to USB > 500 ma power, and each LED need 20 ma x 3 = 60 ma, if I wanna add 67 LEDs, totally need 4000 ma power which is not offered by USB, is there any instructions just tell what to buy, which pin should connect, after that, I can follow the instructions on QMK to config these hardwares.

I searched a whole day by google, and found nothing on making a RGB keyboard using QMK step by step.

Will QMK made these easier to custom a personal keyboard, or is there a RGB per key PCB to buy?

Thanks again bro, lets make QMK greater!

MarsThePlanet, 2019-08-14:
> oh, that's what I want, per key LED.
>
> I saw the configs of QMK files about WS2812, but what should I do before setting these config, how to connect physical WS2812 to pro micro.
>
> Also, due to USB > [...]

The IS31FL3731 will handle the current load for each LED by strobing them at a rate the human eye won't notice.

You will also need to connect your LEDs by charlieplexing them.

The 32u4 and IS31FL3731 can communicate over the I2C protocol, you just need to connect SDA(D1 pin on teensy 2.0) and SCL(D0 pin on teensy 2.0) pins from the 32u4 to IS31FL3731 and provide power over VCC (+5V) and GND, from there on you can follow the RGB Matrix guide from QMK Docs.

You can find more info on adafruit's official site for the driver: https://learn.adafruit.com/i31fl3731-16x9-charliplexed-pwm-led-driver/pinouts

drashna, 2019-08-14:
As somebody that uses a board that uses the WS2812 for the per key RGB, I can tell you that using the ISSI controller is a much better way to go.  

With 67~ LEDs, you'd have to limit the brightness for the WS2812 LEDs, but not with the ISSI based ones.  So you'd get a bright board, with better lighting, that draws significantly less power.

icyhat, 2019-08-22:
thanks all of you, i'll try that

but if i fail, if anyone can help on designing a simple QMK solution of a keyboard, which is create a new keyboard config on qmk, buy what controllers  , how to link pins together, I would pay for the time you helping me made my own keyboard, just let me know, my friends.

drashna, 2019-08-22:
Well, Adafruit has a dev board that you can use to create a prototype to test things out:
https://www.adafruit.com/product/2946

You could also include this in the board design too.  It hooks up via I2C, and you can configure the addresses on the board.  

Additionally, as for LEDs, you have a number of options... but I have a bookmark of these:
https://www.alibaba.com/product-detail/SMD-6028-Smd-Led-RGB-color_60283039151.html?spm=a2700.7724857.normalList.149.715a748fk4G1Vc

These are the type of LEDs that the Planck Light, Planck EZ, Ergodox EZ Glow, and many other per-key rgb boards use.   You need a big cutout, and to use RGB friendly switches.

If you want, head to the QMK Discord, and you can chat with people there. 
https://discord.gg/Uq7gcHh

icyhat, 2019-08-26:
After trying kbfirmware.com, I found it useful for customize a keyboard in physics, is there any official doc for customize physical layout, buying controllers and linking pins together to make RGB and bluetooth work and then I can customize quantom functions in logic via qmk.

kbfirmware.com is good to select controllers and linking pins to keys, but doesn't contain RGB and bluetooth connection in physical, and also not compatible with official tool qmk_configurator.

Guys, is there blogs about how to handwire a custom keyboard, or shall we talk face to face via camera so I can learn how to simple use these tools to build a keyboard rather than dive into non-relative docs.

@drashna is it convenient for you at any time to teach me? I have no other way to give you back except buying your time, or handwire one more keyboard and send to you as a gift?

It looks like this, 
[image]

Like ergodox but more ergonomic, I plan to made it 3D which is fully ergonomic like a mouse, you can hold it and position of every key is adjustable based on the position, length of your finger and hand.

Also once physical position of keys suit your fingers, I wanna make characters layout more ergonomic, like http://mkweb.bcgsc.ca/carpalx/?full_optimization

All this is for the health and productivity of our hands!

Anybody wanna to work with me to make such a keyboard?

drashna, 2019-08-26:
Most of what you want is listed in the QMK documents.

https://docs.qmk.fm/#/hand_wire
That covers a lot of the wiring stuff.

As for RGB, per key or underglow? 
https://docs.qmk.fm/#/feature_rgblight
https://docs.qmk.fm/#/feature_rgb_matrix

For BlueTooth:
https://docs.qmk.fm/#/feature_bluetooth

As for customization:
https://docs.qmk.fm/#/ref_functions
https://docs.qmk.fm/#/custom_quantum_functions

As for a blog, Thomas Baart has a nice blog that goes over a lot of the basics:
https://thomasbaart.nl/

Aside from that, handwiring/designing is outside my realm of experience. 

To be blunt, you may want to actually talk to somebody that designs PCBs, and commission a board from them.   THat may be the simplest option, since they should have a good idea of what you need. 

That said, the best thing to do is to work backwards. Figure out which features are MUST HAVE, and which would be nice.   Then research what it would take to use those.  Like per key RGB needs the rgb matrix, which needs I2C.   And for BlueTooth, that depends on the Controller that you use. But the SPI one would need "SPI".   So you'd want to find a controller that supports both of those, and are supported by the firmware. 

And then determine if you want to use AVR or ARM.  And find a dev board for what you want. 

Getting a breadboard for prototyping would be a very good idea. And you can reuse some (or most) of the parts in the final product. 

And KiiCad is probably what you want to use to design a PCB, if you want to do that. 

That said, this isn't exactly a great place to ask for help with hardware design.  This is really for the firmware side of things.   So again, it may be worth contacting somebody and commissioning a board.

Also, the EEVlog may be a good youtube channel to watch, as they may include some of that information (but I'm not 100% sure about that)

icyhat, 2019-08-26:
@drashna Thanks for so much detailed information.

I am sure I read the docs on docs.qmk.fm, the question is that it only list the supported controllers of main controller, led controller and bluetooth controller, boards, leds types and so on, without how to link them together.

I believe that the qmk is so powerful, but it is tools like kbfirmware.com that enable a man to customize his own keyboard with simple steps.

What a man want is where to buy, how to connect them together.

Here is the early version of my keyboard to test how connect all these things in logic, I even bought ISSI controller to try to add rgb per key, but found nothing on qmk doc to connect them together.

[image]

If qmk is for `develop a completely customizable, powerful, and enjoyable firmware experience for any project`, if I can't even assemble one keyboard using supported objects listed on qmk, how can I then customize it?

Thanks all kindly replies of yours, if anyone wanna receive my ergonomic keyboard when it finished, leave me a address.

Thanks again, I'll try those blogs, : )

closed 2019-08-28 by icyhat