qmk/qmk_firmware #6532: Add support for the Teensy 4.0

Wulfsta, 2019-08-13:
The [Teensy 4.0](https://www.pjrc.com/store/teensy40.html) was recently released, and would be nice to have support for.

## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

zvecr, 2019-08-13:
Support will have to wait till http://www.chibios.org/ support the chip, and given beta hardware has been out a while, there doesn't seem to be much movement on it.

Once that happens, it becomes an issue of upgrading our fork of ChibiOS to include it (this has its own set of issues due to the age of the fork).

Wulfsta, 2021-08-06:
Is the Teensy 4.x supported on the `develop` branch? Seems like it might be as of f533d9e.

Ashtonnot, 2022-02-26:
Is qmk supported on the teensy 4.0???

sigprof, 2022-02-26:
Apparently the only board in QMK using Teensy 4.x is `kinesis/kint41`, which is documented to work with both Teensy 4.0 and 4.1 versions (they use MCUs from the same series which differ mostly in the flash size, and the number of broken out pins is obviously different too, but otherwise the code should be compatible with both of those controllers).

I don't actually have that hardware, but from looking at the code one major missing feature for that MCU family is the EEPROM support (so you won't be able to store persistent settings unless you add an extra I2C or SPI EEPROM chip to your board design). Also there is no hardware-accelerated WS2812 driver for those MCUs that is confirmed to work (there is some chance to make the `spi` driver work, but it probably needs some changes in the driver code, and the `pwm` driver is very STM32-specific), and if you are trying to make a split board, the USART split communication driver may need some adjustments too.

nwaelti, 2022-03-04:
Hello guys, I was wondering if someone managed to get AUDIO_ENABLE = yes support on this board, the Teensy 4.1 on itself works great, but I can't manage to get audio working (pwn software or pwm hardware)... Other than that, QMK works great on the board. Can someone point me in the right direction ?

zvecr, 2022-08-28:
Initial support implemented, community is welcome to iterate on the currently supported low level chibios drivers.

closed 2022-08-28 by zvecr

drGarbinsky, 2022-12-14:
How does one setup a new keyboard project that uses the teensy4.0? I'm getting a bit lost in the documentation.