qmk/qmk_firmware #6534: Issues with Ergodox and Dell Wyse 7030

plattfot, 2019-08-13:
## Describe the Bug
Using a Dell zero client at work with an Ergodox from massdrop (now just drop) using the EZ firmware. The zero client connects to a remote machine about 400km away.

I have had issues with key drops from day one with the zero client. One workaround was to bridge the keyboard to the host. However this caused all my sendstring macros to stop working.
The key drops seems to happen more frequently for capital letters, i.e. pressing shift and some letter.
But this is not all. I flashed my keyboard little more than a month ago and then I also started seeing two types of duplicate artifacts. For example `ee` when pressing `e` and `thisi` when typing `this` i.e. the key repeats after a second key press. Some sort of bouncing issue perhaps? It got a little better after the two PR mentioned in #6269. But it still happens.

## System Information

 - Keyboard: 
   Massdrop second Ergodox drop. Same hardware as the Ergodox EZ.
 - Operating system: 
   Cent6 on host machine, no idea what os the Dell zero client runs.
   Host machine's linux kernel: 2.6.32-431.el6.x86_64
 - AVR GCC version: 
   9.1.0-1
   Firmware built on my laptop at home running Arch Linux.

 - QMK Firmware version: 
   Own fork based on 475d2c0c3

 - Any keyboard related software installed?
   No other keyboard related software.
## Additional Context

This only happens on the Dell zero client. No issues when I tested this on my laptop at home. Nor did I had any issues at work before switching over to the zero client. The workstation I used before the switch was running the same OS as my remote host is currently running.

drashna, 2019-08-22:
Well, could you downgrade your version of AVR GCC.  9.1 is know to have issues, and we don't fully support it yet. 

Also, try compiling with `NKRO_ENABLE = no` in your rules.mk and see if that helps.

plattfot, 2019-08-25:
Rolled back to 8.3, that is noticeable more stable. Still dropping keys but less frequent than when I used 9.1 (also tested 9.2 works better than 9.1 but think 8.3 works better).
I can reliably reproduce the key drops when typing
'This is a test' at moderate speed (around 60wpm)
And it is usually the t in test that gets dropped (I.e This is a est). Same tests worked fine on my laptop.
With 9.1 I see more frequent drops and more prone to bounces. 
I didn't see the weird key bounce though (`this` -> `thisi`) with neither 9.1, 9.2 or 8.3. I'm testing this on a Sunday so maybe it has something to do with network load.

The `NKRO_ENABLE = no` made no difference from what I can tell.
Will test this more with 8.3 and NKRO disabled through out the week and see.

I also tested with an ergotravel and I'm seeing the same keydrop behavior. It's using the same microprocessor though so that was expected. I got a ergodox infinity and a planck at home so I'll see if I can try those two to as well and see if they have the same issue. That will most likely be next weekend.

drashna, 2019-08-30:
Does adding `DEBOUNCE_TYPE = sym_g` help?

plattfot, 2019-09-02:
That seems to help with the bouncing. Just been testing to do some coding now for about an hour and haven't notice and double presses. Or `thisi`. Still dropping keys though.I'll will do some more testing through out the week and see.

I've been testing the NKRO_ENABLE = no and compiled with avr-gcc: 8.3 during the week. I would say it's about the same as what I had before with avr-gcc:9.1. I.e. lot of keydrops and bouncing.
I also brought my ergodox infinity and planck to work today. Seeing the same issues with those as what I'm saw with the ergodox ez and ergotravel last weekend.

I'll report back at the end of this week and see if the bouncing have been solved or not. Looks promising on that front at the moment.

plattfot, 2019-09-08:
Looks like `DEBOUNCE_TYPE = sym_g` fixed the bouncing issues. Haven't experience any double presses or `thisi` after I enabled that.

Still dropping keys though, `return` often becomes `retun`.

plattfot, 2019-12-09:
I haven't had access to a Dell Wyse machine for a while and and will probably take a few months before I have access to them again.

So I'll close this and will reopen if the issue still persist when I do.

closed 2019-12-09 by plattfot