qmk/qmk_firmware #6540: link errors

androclus, 2019-08-15:
I was previously able to compile for my hotdox keyboard a few weeks ago. Now cannot. I am on an OpenSUSE system, and just to be sure, I clone a fresh repo copy:

git clone --recurse-submodules https://github.com/qmk/qmk_firmware.git
cd qmk_firmware

Then I run util/linux_install.sh and this verifies that I have all the correct dependency packages installed on my OpenSUSE/Tumbleweed system.

But when I run

`$ make hotdox:default:dfu`

I get the following:

```Shell
jeff@charlie:~/git/qmk_firmware> make hotdox:default:dfu
QMK Firmware 0.6.449
Making hotdox with keymap default and target dfu

avr-gcc (SUSE Linux) 8.3.1 20190531 [gcc-8-branch revision 271802]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/hotdox/matrix.c                                                                [OK]
Compiling: drivers/avr/i2c_master.c                                                                 [OK]
Compiling: keyboards/hotdox/left.c                                                                  [OK]
Compiling: keyboards/hotdox/hotdox.c                                                                [OK]
Compiling: keyboards/hotdox/keymaps/default/keymap.c                                                [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: quantum/process_keycode/process_unicode.c                                                [OK]
Compiling: quantum/process_keycode/process_unicode_common.c                                         [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
Compiling: tmk_core/common/keyboard.c                                                               [OK]
Compiling: tmk_core/common/action.c                                                                 [OK]
Compiling: tmk_core/common/action_tapping.c                                                         [OK]
Compiling: tmk_core/common/action_macro.c                                                           [OK]
Compiling: tmk_core/common/action_layer.c                                                           [OK]
Compiling: tmk_core/common/action_util.c                                                            [OK]
Compiling: tmk_core/common/print.c                                                                  [OK]
Compiling: tmk_core/common/debug.c                                                                  [OK]
Compiling: tmk_core/common/util.c                                                                   [OK]
Compiling: tmk_core/common/eeconfig.c                                                               [OK]
Compiling: tmk_core/common/report.c                                                                 [OK]
Compiling: tmk_core/common/avr/suspend.c                                                            [OK]
Compiling: tmk_core/common/avr/timer.c                                                              [OK]
Compiling: tmk_core/common/avr/bootloader.c                                                         [OK]
Assembling: tmk_core/common/avr/xprintf.S                                                           [OK]
Compiling: tmk_core/common/bootmagic.c                                                              [OK]
Compiling: tmk_core/common/mousekey.c                                                               [OK]
Compiling: tmk_core/common/backlight.c                                                              [OK]
Compiling: tmk_core/protocol/lufa/lufa.c                                                            [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: tmk_core/protocol/lufa/outputselect.c                                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Linking: .build/hotdox_default.elf                                                                  [ERRORS]
 | 
 | /usr/lib64/gcc/avr/8/ld: warning: -z relro ignored
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/avr/sys-root/lib/avr5/libm.a when searching for -lm
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/avr/sys-root/lib/libm.a when searching for -lm
 | /usr/lib64/gcc/avr/8/ld: cannot find -lm
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/lib64/gcc/avr/8/avr5/libgcc.a when searching for -lgcc
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/lib64/gcc/avr/8/libgcc.a when searching for -lgcc
 | /usr/lib64/gcc/avr/8/ld: cannot find -lgcc
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/avr/sys-root/lib/avr5/libm.a when searching for -lm
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/avr/sys-root/lib/libm.a when searching for -lm
 | /usr/lib64/gcc/avr/8/ld: cannot find -lm
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/avr/sys-root/lib/avr5/libc.a when searching for -lc
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/avr/sys-root/lib/libc.a when searching for -lc
 | /usr/lib64/gcc/avr/8/ld: cannot find -lc
 | /usr/lib64/gcc/avr/8/ld: skipping incompatible /usr/avr/sys-root/lib/avr5/libatmega32u4.a when searching for -latmega32u4
 | /usr/lib64/gcc/avr/8/ld: cannot find -latmega32u4
 | collect2: error: ld returned 1 exit status
 | 
make[1]: *** [tmk_core/rules.mk:300: .build/hotdox_default.elf] Error 1
Make finished with errors
make: *** [Makefile:553: hotdox:default:dfu] Error 1
jeff@charlie:~/git/qmk_firmware> `
```

## Describe the Bug

Although all my dependencies (packages) are satisfied, and although I have changed no keymap files, and although it used to compile fine before, I am now getting linking errors.

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:

/etc/os-release:
```
NAME="openSUSE Tumbleweed"
# VERSION="20190809"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20190809"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20190809"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
LOGO="distributor-logo"
```
uname -a:

```
Linux charlie 5.2.5-1-default #1 SMP Wed Jul 31 10:41:36 UTC 2019 (79b6a9c) x86_64 x86_64 x86_64 GNU/Linux
```

 - AVR GCC version: 
avr-gcc (SUSE Linux) 8.3.1 20190531 [gcc-8-branch revision 271802]
 - ARM GCC version: 
arm-none-eabi-gcc (SUSE Linux) 8.3.1 20190531 [gcc-8-branch revision 271802]
 - QMK Firmware version:
0.6.449
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
None. Let me know if there is any additional information I can supply.

fauxpark, 2019-08-15:
I can reproduce this on a fresh install of Tumbleweed.

Here is what `avr-objdump -x` reports for `libatmega32u4.a`:
```
In archive libatmega32u4.a:

eerd_block.o:     file format elf32-little
rw-r--r-- 0/0    696 Jan  1 10:00 1970 eerd_block.o
architecture: UNKNOWN!, flags 0x00000011:

...
```

Compare with the output on macOS:
```
In archive libatmega32u4.a:

eerd_block.o:     file format elf32-avr
rw-r--r-- 38007580/80    696 Jun  6 09:01 2019 eerd_block.o
architecture: avr:5, flags 0x00000011:

...
```

Similar results for the other files `ld` is complaining about.

Seems like something's up with the OpenSUSE package(s), if this build log is anything to go by: https://build.opensuse.org/package/live_build_log/CrossToolchain:avr/cross-avr-gcc8/openSUSE_Tumbleweed/x86_64

WorldsEndless, 2021-08-13:
I'm having a linking error on Tumbleweed, too. 

```
qmk compile -kb planck/rev6 -km planck-dvp
Ψ Compiling keymap with gmake -j 1 planck/rev6:planck-dvp

QMK Firmware 0.13.25
Making planck/rev6 with keymap planck-dvp

arm-none-eabi-gcc (SUSE Linux) 11.1.1 20210625 [revision 62bbb113ae68a7e724255e17143520735bcb9ec9]
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/command.c                                                                        [OK]
Linking: .build/planck_rev6_planck-dvp.elf                                                          [ERRORS]
 | 
 | /usr/lib64/gcc/arm-none-eabi/11/ld: cannot find -lg_nano
 | /usr/lib64/gcc/arm-none-eabi/11/ld: cannot find -lc_nano
 | collect2: error: ld returned 1 exit status
 | 
gmake[1]: *** [tmk_core/rules.mk:344: .build/planck_rev6_planck-dvp.elf] Error 1
gmake: *** [Makefile:530: planck/rev6:planck-dvp] Error 1
Make finished with errors
```

drashna, 2022-08-14:
Was this still an issue?

androclus, 2022-08-15:
It is not an issue for me (I am OP) but that is only because I sold my HotDox keyboard 3 years ago, a couple of months after posting this bug.  It may or may not still have this problem.

If you need me to try to reload all the libraries and go through the compilation again, nonetheless, let me know. But maybe fauxpark or drashna are still up-to-date and can answer?

fauxpark, 2022-08-15:
We dropped "support" for OpenSUSE a while back: https://github.com/qmk/qmk_firmware/pull/11408

closed 2022-08-28 by zvecr

maiphi, 2022-09-04:
`hotdox:default` builds successfully on openSUSE Tumbleweed. I don't have this keyboard to test the firmware on, but the firmware I built for my Drop CTRL works fine. Needed to install the following dependencies: `cross-arm-none-gcc12 cross-arm-none-newlib-devel cross-avr-gcc7 avrdude avr-libc dfu-programmer dfu-util`.