qmk/qmk_firmware #6549: Layer Toggle sends no input crkbd

leshow, 2019-08-16:
If I take the default board from crkbd and try to add a `LT(LAYER, KC_*)` on the left hand, that key no longer works as a tap key and holding it down does not activate the layer.

The odd part is that the exact same code on the right hand works. here's  sample  [keymap.c](https://github.com/leshow/qmk_firmware/blob/master/keyboards/crkbd/keymaps/leshowmap/keymap.c)

It's just  the default map with a few extra layers. Is there any reason why a LT would work on one hand and not the other? I can rule out any hardware issues I think because I'm able to create the problem on any key if I make it a `LT` but only on the left hand (master hand)

## System Information

 - Keyboard:  crkbd
   - Revision (if applicable):  v2
 - Operating system: linux
 - AVR GCC version: 8.3.0
 - ARM GCC version: 8.3.0
 - QMK Firmware version: master
 - Any keyboard related software installed? 
   none

raetiacorvus, 2019-08-17:
`LT(KC_F, ARRW)` and `LT(KC_V, NUMB)` shouldn't those be `LT(ARRW, KC_F)` and `LT(NUMB, KC_V)` ?

leshow, 2019-08-17:
You're absolutely right, thanks!

closed 2019-08-17 by leshow