qmk/qmk_firmware #6551: massdrop led does not turn off using #define RGBLIGHT_SLEEP

pratyushtewari, 2019-08-16:
I wanted the LED on massdrop CTRL keyboard to turn off when the computer is sleeping. I tried adding `#define RGBLIGHT_SLEEP` to the `keyboards/massdrop/ctrl/config.h` file but it does not turn off the led.

## Describe the Bug
`#define RGBLIGHT_SLEEP` does not work on massdrop CTRL keyboard

## System Information

 - Keyboard:  Massdrop CTRL
 - Operating system: Mac osX 10.14.6 (18G87) Mojave
 - AVR GCC version:  8.3.0
 - ARM GCC version: 7.3.1 20180622
 - QMK Firmware version: 0.6.103

fauxpark, 2019-08-16:
The CTRL uses the RGB Matrix feature, not RGB Light (these are different things). It doesn't look like there is an equivalent define for the former, but this section of the docs may help: https://docs.qmk.fm/#/feature_rgb_matrix?id=suspended-state
You can add the `_user()` versions of those functions in your keymap, and it should achieve the same result.

pratyushtewari, 2019-08-16:
Hi @fauxpark I tried this, I must be doing some mistake.

See: https://github.com/pratyushtewari/qmk_firmware/commit/9df4f16bf7bb7012524c123188da66e2cbc90801#diff-b7344527f4c3c147d8aa7acbb45611aaR63

After suspend the keyboard is still lit like 4th of July.

fauxpark, 2019-08-16:
Remove the `_keymap` part from the function names 😉

pratyushtewari, 2019-08-16:
Awesome! that worked.
Is there any way to turn off the led after I  turn off the display using the 
Shit + Ctrl + Power key? 

after I run the macro
https://github.com/pratyushtewari/qmk_firmware/commit/9df4f16bf7bb7012524c123188da66e2cbc90801#diff-b7344527f4c3c147d8aa7acbb45611aaR95

how would I wake up? in that scenario. Can I wake up the led when I press the key again?

Thanks for the help. This is making my day.

fauxpark, 2019-08-16:
I presume macOS doesn't place the keyboard into the suspend state when you're just sleeping the display. So, there is probably no way to get that information out of the host.

pratyushtewari, 2019-08-16:
I was hoping we can call the `rgb_matrix_set_suspend_state(true);` after running the macro send string function. I’ll try that. Question is how do I call `rgb_matrix_set_suspend_state(false);` again?

fauxpark, 2019-08-16:
That's the problem; you can't know when to call the function because the host doesn't tell the keyboard. Technically, you could set an "is display sleeping" variable in the `SND_SLEEP` case, and on every keypress check if it is `true` or `1` etc. but there is a decent potential for this variable becoming out of sync with the actual state of the display. So I don't recommend it unless you're fine with that possibly occurring.

pratyushtewari, 2019-08-16:
Yeah. I'll keep at this. If I can solve this, it would be awesome. Thanks for all the help. I'll close the issue now.

closed 2019-08-16 by pratyushtewari