qmk/qmk_firmware #6557: fast Tapped Key not registering with RETRO_TAPPING and TAPPING_TERM < 100

raetiacorvus, 2019-08-17:
## Describe the Bug

Hi in my layout i use the following two LT defines:
```
#define ST(LAY) LT(LAY, KC_SPACE)
#define NAV_ESC LT(NAV, KC_ESC)
```

When using them with `RETRO_TAPPING` and `TAPPING_TERM 100` or smaller i noticed that space or esc would not register sometimes even when i clearly pressed and released the corresponding key. After some testing i could easily reproduce this by tapping an LT Key quickly after any other key on my default layer. e.g. <kbd>A</kbd> then <kbd>Space/Layer</kbd> will result in just 'A' rather then 'a ' with a space after it. It seems so be easier to reproduce this by pressing <kbd>Space/Layer</kbd> while <kbd>A</kbd> is still pressed and then release <kbd>A</kbd> first and then <kbd>Space/Layer</kbd>.

## Debugging for the example
### Matrix
r/c for <kbd>A</kbd> is 02/1
r/c for <kbd>Space/Layer</kbd> is 00/7
Case 1 of missing space
```
r/c 0123456789ABCDEF
00: 0000000000000000
02: 0100000000000000

r/c 0123456789ABCDEF
00: 0000001000000000
02: 0100000000000000

r/c 0123456789ABCDEF
00: 0000001000000000
02: 0000000000000000

r/c 0123456789ABCDEF
00: 0000000000000000
02: 0000000000000000
```
Case 2 of missing space
```
r/c 0123456789ABCDEF
00: 0000000000000000
02: 0100000000000000

r/c 0123456789ABCDEF
00: 0000001000000000
02: 0000000000000000

r/c 0123456789ABCDEF
00: 0000001000000000
02: 0000000000000000

r/c 0123456789ABCDEF
00: 0000000000000000
02: 0000000000000000
```
### keyboard report
Missing space
```
keyboard_report: 00 00 04 00 00 00 00 00 
keyboard_report: 00 00 04 00 00 00 00 00 
keyboard_report: 00 00 00 00 00 00 00 00 
keyboard_report: 00 00 00 00 00 00 00 00
```
Working Space
```
keyboard_report: 00 00 04 00 00 00 00 00 
keyboard_report: 00 00 00 00 00 00 00 00 
keyboard_report: 00 00 00 00 00 00 00 00 
keyboard_report: 00 00 00 00 00 00 00 00 
keyboard_report: 00 00 2C 00 00 00 00 00 
keyboard_report: 00 00 00 00 00 00 00 00
```

## System Information

 - Keyboard: Katana60
 - Operating system: Arch Linux
 - AVR GCC version: 9.2
 - QMK Firmware version: 0.6.451

ales, 2019-08-18:
Can replicate, have the same issue.

```
#undef TAPPING_TERM
#define TAPPING_TERM 10

#define RETRO_TAPPING
```

Keyboard: Plaid (ATMEGA328P)
OS: MacOS
avr-gcc (GCC) 8.3.0
QMK Firmware 0.6.453

ales, 2019-08-18:
so possibly fixed it. at least it works for me and see no problem so far. attaching patch file

[0001-fix-to-6557.patch.txt](https://github.com/qmk/qmk_firmware/files/3513660/0001-fix-to-6557.patch.txt)

raetiacorvus, 2019-08-21:
@ales thank you very much, your patch fixes the issue for me. I have been using it for the last 3 days and at least with my keymap/typing style i wasn't able to notice any issues with it.

drashna, 2019-08-22:
@ales  I think you have a PR for this, as well?

closed 2019-12-21 by stale[bot]

AckslD, 2021-05-14:
Was this fixed actually? I'm experiencing the same issue

rautesamtr, 2021-05-14:
Not fixed as far as i can tell. Here's a rebased version of @ales patch. It still works.
[0001-Fix-6557.patch.txt](https://github.com/qmk/qmk_firmware/files/6479484/0001-Fix-6557.patch.txt)

AckslD, 2021-05-14:
Thanks @rautesamtr, patch works for me :+1:. I opened a PR with the patch (https://github.com/qmk/qmk_firmware/pull/12899)

AckslD, 2021-06-09:
@rautesamtr I'm using the [workaround](https://github.com/qmk/qmk_firmware/pull/1359#issuecomment-522360151) by @ales 
```
#undef TAPPING_TERM
#define TAPPING_TERM 10

#define RETRO_TAPPING
```
with the patch.

One thing that I'm noticing is that if I have let's say the following to buttons:
```
a = LGUI_T(KC_ESC)
b = LT(_LOWER, KC_ENT)
```
and I want to press `GUI+ENT`, which should be done by holding `a` and pressing `b`, this doesn't work and I have to press `b` twice.

There seem to be something that needs to be activated first, ie I can hold `a` press any other button and then the sequence is correctly triggered when pressing `b` the first time.

Any idea why this might be?

AckslD, 2021-07-04:
Anyone stumbling on this, the above issue is solved in #12899 for at least these settings