qmk/qmk_firmware #6567: OSM key not registered when other key is pressed

roggan87, 2019-08-18:
Thanks for the excellent work with QMK, it's such a blast!

I've set up shift like an OSM (one shot modifier) and when typing fast enough I sometimes press it before I've released the previous key. The expected behavior would've been that the OSM was registered and applied to the next keypress, not omitted. Would it be possible to achieve such behavior? Are there any obvious drawbacks or unwanted side effects?

Note: If the OSM is pressed _and released_ before releasing the previous key, it is registered as expected. I'll try to illustrate (just to be obvious)

1. A down, OSM down, A up, OSM up, B, results in ab
2. A down, OSM down, OSM up, A up, B, results in aB

I would've expected both scenarios to result in aB. I might be able to help fix it if pointed in the right direction in the code base. Fixing this issue would result in so much smoother camelCasing and PascalCasing.

Thanks again, you're awesome!

drashna, 2019-08-18:
Which keyboard is this for? 

Because I can't reproduce this on my ergodox ez, where I use OSM extensively.

roggan87, 2019-08-19:
That's interesting! This is on a Dactyl.

roggan87, 2019-08-19:
Some more investigation shows that OSL does not have this issue. I've tried OSM on both the microcontroller side and the IO expander side, both are having this issue.

drashna, 2019-09-26:
That's is very weird, and I suspect that this is a low level issue. 

I suspect that this is an edge case in the handling, but I'm not entirely sure where.  

for reference, the actual code for this is here:
https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/action.c#L252-L293

closed 2022-08-28 by zvecr