qmk/qmk_firmware #6571: OSM staying on for two rolling keypresses

pauleikis, 2019-08-19:
## Describe the Bug

After pressing OSM and then typing two letters in quick succession, both are affected, i.e. typing rolling "The" as reported by Karabiner
Keypresses: OSM down, OSM up, t down, h down, t up, h up
Keycodes sent: shift down, t down, h down, shift up, t up, h up
Expected keycodes: shift down, t down, shift up, h down, t up, h up

## System Information

 - Keyboard:  Planck EZ
 - Operating system: Mac OS
 - AVR GCC version: will update later

 - ARM GCC version: will update later

 - QMK Firmware version: recent, will update later

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ x] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2019-08-22:
Unfortunately, I'm not able to replicate this behavior, so it may be a timing issue on the system. 

That said, there are some possible debugging options that may make it easier to fix. 
https://docs.qmk.fm/#/feature_advanced_keycodes?id=callbacks

pauleikis, 2019-08-22:
Apparently that was an issue with Karabiner, quiting it resolved the issue and now one shot mods are working fine in this regard. Thanks for your time in checking this!

closed 2019-08-22 by pauleikis

drashna, 2019-08-22:
lol, well, glad that you found the source of the issue, at least!

pauleikis, 2019-08-23:
Just to clarify, this was just vanilla Karabiner, without any magic presets or functions, I was using it to debug keycodes for unrelated issues. Sounds like a bug in Karabiner