qmk/qmk_firmware #6572: Key MAGIC_TOGLE_ALT_GUI has no effect.

Lithium768, 2019-08-19:
Start at line 556 of `quantum/quantum.c`
```
    case MAGIC_SWAP_CONTROL_CAPSLOCK ... MAGIC_TOGGLE_NKRO:
      if (record->event.pressed) {
        // MAGIC actions (BOOTMAGIC without the boot)
        if (!eeconfig_is_enabled()) {
            eeconfig_init();
        }
        /* keymap config */
        keymap_config.raw = eeconfig_read_keymap();
        switch (keycode)
        {
/* omitted 576~637 */
          case MAGIC_TOGGLE_ALT_GUI:
            keymap_config.swap_lalt_lgui = !keymap_config.swap_lalt_lgui;
            keymap_config.swap_ralt_rgui = !keymap_config.swap_ralt_rgui;
            #ifdef AUDIO_ENABLE
              if (keymap_config.swap_ralt_rgui) {
                PLAY_SONG(ag_swap_song);
              } else {
                PLAY_SONG(ag_norm_song);
              }
            #endif
            break;
```
`case MAGIC_SWAP_CONTROL_CAPSLOCK ... MAGIC_TOGGLE_NKRO:` cannot capture `MAGIC_TOGGLE_ALT_GUI`, according to
`quantum/quantum_keycodes.h` line 99~118
```
    MAGIC_SWAP_CONTROL_CAPSLOCK,
    MAGIC_CAPSLOCK_TO_CONTROL,
    MAGIC_SWAP_LALT_LGUI,
    MAGIC_SWAP_RALT_RGUI,
    MAGIC_NO_GUI,
    MAGIC_SWAP_GRAVE_ESC,
    MAGIC_SWAP_BACKSLASH_BACKSPACE,
    MAGIC_HOST_NKRO,
    MAGIC_SWAP_ALT_GUI,
    MAGIC_UNSWAP_CONTROL_CAPSLOCK,
    MAGIC_UNCAPSLOCK_TO_CONTROL,
    MAGIC_UNSWAP_LALT_LGUI,
    MAGIC_UNSWAP_RALT_RGUI,
    MAGIC_UNNO_GUI,
    MAGIC_UNSWAP_GRAVE_ESC,
    MAGIC_UNSWAP_BACKSLASH_BACKSPACE,
    MAGIC_UNHOST_NKRO,
    MAGIC_UNSWAP_ALT_GUI,
    MAGIC_TOGGLE_NKRO,
    MAGIC_TOGGLE_ALT_GUI,
```
This makes `MAGIC_TOGGLE_ALT_GUI` key ineffective.

fauxpark, 2019-08-19:
Good catch! I made a PR to fix this, see above :)

drashna, 2019-08-22:
This should be fixed now, and now we have Ctl - GUI swap, as well.

closed 2019-08-22 by drashna