qmk/qmk_firmware #6577: FR: Auto shift layers

bsurmanski, 2019-08-19:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

## Description
Auto shift (capitalize letters when holding the key) is a very useful feature, but I think it could be even more useful if there was the option to insert characters from a specific layer.

Use cases: convenient numberpad, quick function keys (F4 key), use fewer keys

My motivation: I get RSI pain in my right wrist. I have an ergodox, and I've been experimenting with one-handed typing. So far, I have a layer with all letters mirrored (left hand becomes right hand). I have a layer toggle on the thumb. But of course, this requires a lot of layer toggles (multiple per word). I was thinking the ideal way for it to work would be similar to auto-shift, hold a letter to insert the equivalent swap handed letter. For example, to type "hello" i would type "GeSSW", where each capital letter is autoshifted to a "mirrored right hand" layer.

Some concerns: this likely would have cornercases with modifier keys like CTRL or layer toggles. But I think the same corner cases already exists with other layer toggle keys.

drashna, 2019-08-19:
Two things you may want to look into actually: 
* [One Shot Keys](https://docs.qmk.fm/#/feature_advanced_keycodes?id=one-shot-keys)
* [Swap Hands](https://docs.qmk.fm/#/feature_swap_hands)

Both of these features may help with what you want

bsurmanski, 2019-08-20:
Thanks, the Swap Hands feature is effectively what I am looking for, I don't know how I missed it... I think there is still potential value in shifting layers, but I will close this FR for now.

closed 2019-08-20 by bsurmanski

bsurmanski, 2019-08-21:
After looking further into this, the Swap Hands feature requires a modifier key, and one shots keys also use modifiers. Both of these are functionally equivalent to what I'm using now (SwapHanded layer with OSL). 

What I was thinking of was to swap hands on a individual key basis by holding the key, skipping the need for any specific modifier keys.