qmk/qmk_firmware #6586: Planck RGB lighting question.

aprekates, 2019-08-21:
Can i configure a keyboard so that in  a 'virtual layer' i want certain keys lit in certain color and in the same layer to be able enable some preset rgb effects?

Also if in the a layer i have a key set to act as an RGB on/off , and in the same layer i've set

drashna, 2019-08-22:
Yes.

> Can i configure a keyboard so that in a 'virtual layer' i want certain keys lit in certain color
For this, you want `rgb_matrix_indicators_user`, and to call `rgb_matrix_set_color( i, r, g, b)`

> in the same layer to be able enable some preset rgb effects?

That's more complicated, but it's doable. 

>Also if in the a layer i have a key set to act as an RGB on/off , and in the same layer i've set a color for that key could that inhibit its function as an rgb on/off switch?

You mean light up the rgb toggle key based on the the current state?  

If so, then yeah, you can definitely do that. 

You'd need to add: 
```c
extern rgb_matrix_config_t rgb_matrix_config;
```

And then you can check to see if it's enabled by calling:
```c
if (rgb_matrix_config.enable) {
    // do the thing if enabled
} else {
    // do the thing if not enabled
}
```

That said, if this is the planck ez, the Oryx configurator may be able to handle most of that for you.

aprekates, 2019-08-22:
I tried the above in the Oryx with little luck. is there a git page for oryx ?

drashna, 2019-08-22:
For the configurator itself, no.  But you can download the source for the keymap, and see what it does.

drashna, 2019-09-26:
Have you had any luck with this?

closed 2021-01-12 by drashna