qmk/qmk_firmware #6587: Issue compiling qmk for ymdk_np21

dontwatchthisspace, 2019-08-21:
I'm experiencing a similar issue to #3718, but haven't been able to resolve this with the advice provided, so I'm hoping someone can help.

So to begin, I'm using Windows 10, running msys2, and following the [QMK Getting Started](https://docs.qmk.fm/#/newbs_getting_started). I have installed qmk and trying to set the default keymap running the following as per the README.md of [ymdk_np21](https://github.com/qmk/qmk_firmware/tree/master/keyboards/ymdk_np21 ) using the code below:

`make ymdk_np21:default:program`

I then get the following error message:

Traceback (most recent call last):
  File "./util/atmega32a_program.py", line 23, in <module>
    import usb
ModuleNotFoundError: No module named 'usb'
make[1]: *** [tmk_core/avr.mk:125: program] Error 1
Make finished with errors
make: *** [Makefile:553: ymdk_np21:default:program] Error 1

@fauxpark recommended installing pyusb, or possibly using the details in the [README.cd of ymd75](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ymd75/README.md) but the command he suggested `pip install pyusb` doesn't work in msys2. I've installed python2 using `pacman -S python2`, but this doesn't seem to include this pyusb part?

The ymd75 README.md also recommended trying to use bootloader to load the hex file, but as it's Windows, HIDBootFlash. I've downloaded this, but it doesn't seem to launch.

I've then just run `make ymdk_np21:default` and this creates the .hex file. I just seem to be stuck at loading it into the keyboard!

Does anyone have any suggestions?

fauxpark, 2019-08-21:
`pip3 install pyusb` should do it, if you've run the `qmk_install.sh` script recently - it now installs Python 3 along with everything else.

drashna, 2019-08-22:
@fauxpark It still won't help.  pyusb isn't supported on MSYS2, IIRC.

So you'd need to use QMK Toolbox or BootloadHID to flash it.

dontwatchthisspace, 2019-08-22:
Okay, so using QMK Toolbox, I have managed to flash the keyboard with ymdk_np21_default.hex, and everything seemed to be fine However, now it doesn't type anything at all. So I tried to then flash it again, and when I try to connect it in bootloader mode, it just keeps connecting and disconnecting in a loop.

dontwatchthisspace, 2019-08-28:
I have been able to resolve the issue with the USB disconnect loop by uninstalling all of the keyboards in Device Manager and letting Windows re-install the USB drivers. I have then managed to get HIDBootFlash working to flash the board with the default keyset generate by qmk_toolbox. 

However, the keyboard still doesn't type after this. Not sure if the original issue is therefore considered closed and I should open a new issue, or if I should leave this open here asking for further advice.

dontwatchthisspace, 2019-09-06:
Issue was with the built keyset due to an issue with avr-gcc. After reinstalling msys2 and qmk again, this fixed the issue and keyboard is now working. Issue resolved.

closed 2019-09-06 by dontwatchthisspace