qmk/qmk_firmware #6589: Ctrl and OS keys swapped in Ergodox EZ

kyren, 2019-08-22:
Hey, when building and flashing 1c805b3d008a98a5a15112b0655ba14b42dbae67 for an ergodox ez my OS and CTRL keys are swapped.  I tried macos and linux and observed the same behavior.

If I swap to commit 6a79b05cf02d5a6f07b17474344e272b44c77988 (right before PR #6110) the problem goes away

## System Information

 - Keyboard:  ergodox ez
   - Revision (if applicable):  
 - Operating system: macos and linux (nixos)
 - AVR GCC version: avr-gcc (GCC) 7.4.0

 - ARM GCC version: 
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

 - QMK Firmware version:
1c805b3d008a98a5a15112b0655ba14b42dbae67

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Firmware is built using the toolchain from shell.nix

drashna, 2019-08-22:
Resetting the EEPROM will fix the issue.

kyren, 2019-08-22:
Oh, okay that makes sense.  Sorry if this is a basic question but I can't quite figure out how to reset the eeprom on an ergodox without changing my firmware to give it the ability to do that from a keypress?  I'm using the QMK toolbox GUI and there's a "reset eeprom" button but it doesn't appear to do anything.  Is there a way to do this from cli using dfu-util?

kyren, 2019-08-22:
I tried to use the method you mentioned here: https://github.com/zsa/wally/issues/7 to reset the eeprom, but after doing this the issue is still there, so I'm not sure if I was successful?

drashna, 2019-08-22:
If you have bootmagic enabled, you can Space + backspace when plugging it in. 
If you have bootmagic lite enabled, that will do it. 

Otherwise, if you add `EEP_RST` to the board, or flash this hex, and then reflash yours:
https://github.com/qmk/qmk_toolbox/blob/ef12804cc99bd133f77650865a46f971c8e2f544/common/atmega32u4_eeprom_reset.hex

drashna, 2019-08-22:
Try this hex:
https://drive.google.com/open?id=1jHLYA1gwx_umdJxiySzDz9d-K9XFjh4a

kyren, 2019-08-22:
Yep, flashing that hex did the trick, thank you so much!

Should I close this issue or let #6590 close it?

drashna, 2019-08-22:
Let the PR close it.

And actually, I'm going to pin this for now.

drashna, 2019-08-22:
Specifically, the reason that this has happened is that a new block is being used for the EEPROM. 
Unfortunately, because it's a block that wasn't used previously, it has not been zeroed out properly, so it has a random value.  

This means that the Ctrl-GUI swap may be enabled, because the bit used for it was on, randomly. 

We are looking at both short term and long term solutions for this issue, so that this doesn't happen again, in the future.

Wilba6582, 2019-08-24:
thanks for the heads up @drashna 

🙄

drashna, 2019-08-24:
@Wilba6582 sorry about the collision!

bcat, 2019-08-25:
Had the same issue on a Via-supporting board (Instant60) and the config.h workaround above fixes it. For now I've added this to my personal config.h in my userspace, but should I also send a pull request to fix this in the keyboard's config.h instead? Or is someone doing a batch update of all the Via boards?

Wilba6582, 2019-08-25:
@bcat I will update the VIA keyboards

ilpsxnus, 2019-09-02:
As a Windows only user I was unable to reset the EEPROM through the QMK Toolbox or by activating the mapped "EEPROM Reset" key combo on my keyboard. It's probably not overwriting the specific memory address which holds the swap value. What worked was assigning "MAGIC_UNSWAP_LCTL_LGUI" to a key then activating it, but this needs to be done each time the keyboard is plugged in.

yanfali, 2019-09-03:
There's a bug in the way toolbox erase's eeprom with atmel-dfu right now, it needs `dfu-programmer atmega32u4 erase` before running the erase command.

orbitingorca, 2019-09-03:
Seeing this issue on zeal60. My hacky fix was to comment out code in quantum/keycode_config.c.

drashna, 2019-09-03:
@orbitingorca that's because there is a collision.  The block that the CTRL GUI swap is in is using the same block that VIA uses.   If you add the block I've posted above into your personal keymap's config.h, it will fix the issue without disturbing anything else.

Pimentoso, 2019-09-05:
Issue was happening with my Pro Micro based keyboard (redox rev1)
Solved the issue with
```
cd qmk_firmware
git pull
make clean
make git-submodule
```

and then reflashing my keymap with make redox/rev1:xxxxx:avrdude

stanrc85, 2019-09-06:
My ESC key no longer works on the first layer, seems to have just started since this issue came up and was resolved. No matter what keycode I try the ESC key doesn't work on the default layer, if I switch layers then it works fine.

drashna, 2019-09-06:
@stanrc85 that's really odd.  And from what you describe, I'm not sure that it's related to this issue (but it's hard to tell). 

I can't reproduce your issue, locally.  Are you using grave escape?

stanrc85, 2019-09-06:
I usually have a custom tap dance there, but also tried with grave escape (shift+esc, gui+esc didn't work either), regular escape and even a KC_1 just to test.  Just as reminder its a HS60/v2 board since that usually seems to make a difference. 

I didn't know if it might be related to EEPROM stuff moving around since I only noticed it around the same time.

stanrc85, 2019-09-15:
@Wilba6582 is this still being worked on to fix the via boards?

drashna, 2019-09-18:
This should be fixed now.  Closing.

closed 2019-09-18 by drashna

makenova, 2019-12-28:
I ran into this issue, left Ctrl and the left OS key were swapped, but I am using an instant60.
My local repo was a bit stale but even after rebasing upstream, the issue persisted. Trying to reset the EEPROM with QMK Toolbox didn't work. I eventually mapped "reset eeprom" to a layer and that worked.
I don't want to reopen this issue, I've resolved it for myself.
I'm simply adding this in case someone else runs into this issue or if the repo owners want more info to fix this.

arialdomartini, 2022-10-28:
This happened to me today on DZ-60, out of the blue.

* `MT(MOD_LCTL, KC_D)` and `MT(MOD_RCTL, KC_K)` just acted as `KC_D` and `KC_K` (`MT` being ignored)
* `MT(MOD_LGUI, ...` behaved as `MT(MOD_LCTL, ..`.

I confirm that the following fixed the problem:

* assign `QK_CLEAR_EEPROM` to a random key
* flash the firmware
* press the key programmed with `QK_CLEAR_EEPROM`: this re-initializes the keyboard's EEPROM (persistent memory)
* flash back the original desired keymap

That worked for me. 🥳