qmk/qmk_firmware #6594: split keyboard halves in different state, can't reset

leshow, 2019-08-23:
I'm using the crkbd. 

I hit the RGB_TOG key as the computer was going in to suspend, somehow that caused the lights to go off on the right half of my crkbd but not the left. Hitting toggle again just turns the left half on and the right half off, reflashing doesn't change the default state of the LEDs, even flashing a different layout.

What can be done? AFAIK there isn't an RGB_ON (https://beta.docs.qmk.fm/features/feature_rgblight#keycodes)

leshow, 2019-08-23:
I was able to flash the `RGB_TOG` to one half then plug in that half, and get them both back in the same state, but perhaps there should be an `RGB_ON` / `OFF`. There's a `BL_ON`/`BL_OFF` as well as `BL_TOGGLE`

drashna, 2019-08-23:
The simple option would be to reset the eeprom on both halves

closed 2019-11-23 by leshow