qmk/qmk_firmware #6602: LEDs for layers not showing on Redox

pixelbreaker, 2019-08-24:
Since creating my own custom layout for my Redox wireless, the layer status indicator LEDs have completely stopped working, has anyone else come across this issue?

noroadsleft, 2019-08-24:
@mattdibi Any ideas?

pixelbreaker, 2019-08-24:
I've figured this out, don't worry. I was using the QMK configurator to compile, but now I download the keymaps into my own layout for the redox_w and match the layer indexes in the `matrix_scan_user` switch statement.

It's a shame it doesn't work out of the box with the QMK configurator though 😢

closed 2019-08-24 by pixelbreaker

noroadsleft, 2019-08-24:
Yeah, QMK Configurator doesn't quite have any support for advanced features like indicator LEDs or custom macros. Its API would need changes for that to happen. It's on our wish list, but no timeline as of yet.