qmk/qmk_firmware #6603: Atreus Workman PCBDOWN question

kensanata, 2019-08-24:
I have a Workman layout for the old Atreus keyboard (PCBDOWN=yes to prevent a mirrored layout). My code is based on a code base that's a year old (my [workman branch](https://github.com/kensanata/qmk_firmware/tree/workman)). When I tried to use it with an updated code base, it no longer works as expected (my [workman2 branch](https://github.com/kensanata/qmk_firmware/tree/workman2)):

1. the key I assigned KC_LGUI acts as Control key
2. the key I assigned KC_LCTR acts as Meta key
3. the key I assigned KC_LALT acts as GUI key

I've seen talk about "new code conventions" (https://github.com/qmk/qmk_firmware/pull/5849) but I have no idea what I'm looking for. What do I need to do? Do I need to base my code on the future_unformatted branch?

noroadsleft, 2019-08-24:
That code conventions PR isn't relevant to your current issue, because those changes aren't part of QMK's master branch.

The cleanest way to update is to rebase your workman branch on top of the current QMK master branch. To do this through the command-line (I apologize if you're using a GUI program, as I don't use one so can't walk you through that), run:

```shell
git fetch upstream master:qmk_master  # Requires https://github.com/qmk/qmk_firmware.git to be set as a remote named upstream
git checkout workman
git rebase qmk_master
```

This series of commands fetches the commits from QMK's `master` branch and saves them to a local branch named `qmk_master` (I did this because I snooped your GitHub fork and found that you have a commit on your `master` branch that isn't on QMK's, so we can't use your `master` branch directly), checks out your `workman` branch, and then takes the commits on `workman` and begins applying them on top of `qmk_master`.

Right off the bat, you'll get a merge conflict, which is what happens when two commits modify the same part of a file concurrently. The merge conflicts will be in `keyboards/atreus/readme.md` and `keyboards/atreus/rules.mk`.

If you open the readme in a text editor, you'll find this at the bottom:

```
<<<<<<< HEAD
See [build environment setup](https://docs.qmk.fm/#/getting_started_build_tools), then the [make instructions](https://docs.qmk.fm/#/getting_started_make_guide) for more information.
=======
If your keyboard layout is a mirror image of what you expected (i.e. you do not get QWERTY on the left but YTREWQ on the right), then you have an A-Star powered Atreus (older than March 2016) with PCB labels facing *down* instead of up. Specify that by adding `PCBDOWN=yes` to your `make` commands.

See [build environment setup](https://docs.qmk.fm/build_environment_setup.html) then the [make instructions](https://docs.qmk.fm/make_instructions.html) for more information.
>>>>>>> Support PCBDOWN variants for Atreus keyboard
```

This is the merge conflict. The part between `<<<<<<< HEAD` and `=======` was changed on your branch, and the part between `=======` and `>>>>>>> Support PCBDOWN variants for Atreus keyboard` was changed on QMK's `master` branch. Delete everything from `<<<<<<< HEAD` to `=======`, and the line `>>>>>>> Support PCBDOWN variants for Atreus keyboard`, and save the file. Do the same for `rules.mk`. `git add` both files. then run `git rebase --continue`. This synchronizes the changes from your fork and branch to those made on QMK's `master` branch, as if you had originally made your changes on top of QMK's current `master` branch.

We have a document detailed Git best practices as they pertain to QMK here: https://docs.qmk.fm/#/newbs_best_practices

noroadsleft, 2019-08-24:
Regarding your Ctrl, GUI and Alt keys being scrambled, resetting your EEPROM should fix that. You may need to update your workman branch first and then flash from a .hex compiled from the updated branch. There's a post with solutions in a pinned issue [here](https://github.com/qmk/qmk_firmware/issues/6589#issuecomment-524042457).

kensanata, 2019-08-25:
Thank you very much for the long git explanation. Quickly flashing that other hex file and then reflashing my own fixed the issue. Thanks! I'll get my git into shape and see about contributing that keymap now that it works as intended.

closed 2019-08-25 by kensanata