qmk/qmk_firmware #6634: .hex instead of .bin?

zezoca1, 2019-08-29:
I was flashing my keyboard (dz60rgb) and when I open the softeare and tried to flash it said "! Only firmware files in .bin format can be flashed with dfu-util!" but everywhere I go download my firmwares it always downloads as hex files. Is there a way around this? Thanks.

zvecr, 2019-08-29:
The firmware for a dz60rgb is located here <https://github.com/qmk/qmk_firmware/tree/master/keyboards/dztech/dz60rgb>, and precompiled here <https://qmk.fm/compiled/dztech_dz60rgb_default.bin>. The file you were trying to flash is for the regular AVR based dz60 which is why you were having issues.

yanfali, 2019-08-30:
@zezoca1 this doesn't really seem like a bug, please close the issue if it's resolved. tl;dr your keyboard is ARM based not AVR so it doesn't use HEX files, it uses BIN.

closed 2019-08-30 by zezoca1