qmk/qmk_firmware #6650: Different OLED font sizes

burlapsax, 2019-08-31:
## Describe the Bug

Is it possible to use larger than 6x8 fonts on the 128x32 SSD1306 display using the [standalone OLED library](https://github.com/qmk/qmk_firmware/pull/5288)? I tried creating a double-sized font file, converting it to hex and swapping it into glcdfont.c, changing OLED_FONT_WIDTH and _HEIGHT to 12 and 16, and building again. The output looked like I got the full width of the 12x16 characters – but only the top halves of them. The line breaks were still at the same height as with the 6x8 font, so each line overwrites the lower half of the preceding one.

What do I need to change to get newlines to honor the font height (or, more generally, to get a larger font size than 6x8 to work with this OLED driver)?

## System Information

 - Keyboard:  1987 Apple Desktop Bus keyboard -> Arduino Pro Micro (custom build of keyboards/converter/adb-usb with OLED_DRIVER_ENABLE) + SSD1306 128x32 OLED display
   - Revision (if applicable):  
 - Operating system: macOS 10.14.6
 - AVR GCC version: avr-gcc (GCC) 9.1.0

 - ARM GCC version: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 - QMK Firmware version: forked from 0.6.459 // 43b030997 Add 2015 revision of Pegasus Hoof to QMK (#6595)

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [X] Karabiner
   - [ ] Other:

## Photo of the current setup
[image]

drashna, 2019-09-26:
Tagging @XScorpion2 as they're the author of the OLED Driver feature, for help here. 

And very neat/nice mod!

XScorpion2, 2019-09-26:
Adding tall font support is on my list of todos. I just haven't gotten around to it yet. 

> save image and convert back to hex
Depending on how you convert to hex, this result may or may not be correct.

su8044, 2020-01-19:
Did you ever find a solution? I'm having the same issue with a 128x64 display and the don't is just too small and there is too much screen wasted

brickbots, 2020-03-12:
Hi @XScorpion2!  I'm happy to take this on as I'm interested in using larger font sizes myself.  I've taken a look at the current code and I believe it shouldn't be too hard or memory intensive to support fonts that are multiples of 8 pixels tall with fixed (but arbitrary) width.  I'm thinking 2x, 3x, 4x height to start.

One thing I've run into playing with this is the font format currently used.  The comment in the font file:
// Helidox 8x6 font with QMK Firmware Logo
// Online editor: http://teripom.x0.com/

Refers to an online editor which seems defunct.  I've found this one, which seems fairly well put together:
http://oleddisplay.squix.ch/#/home

But generates variable width font files with a more complicated encoding/more overhead than we probably want here.  In the future we can implement a fully raster system for the oled which is compatible with some of the existing GFX libs, but I think it will be difficult to get something this complex to fit memory along with all the core features.

**Are we okay using our own font file format with no user-friendly tool to create them?**  
If so, we can slightly alter the current format to support taller fonts and include a few different font files that people can use.  This will allow us to keep the implementation fairly lightweight, but provide a few font options in various sizes, with the ability to mix them on screen at once.

**Here is what I propose:**
* Add two bytes to the start of the font array, one for width, and one for height.  This will allow predictable indexing into the font array to render characters without needing a lookup table.
* Create a number of font files supporting various sizes/styles
* Create specific font files for the glyphs currently used for logos and such
* Come up with some sort of clever way to allow multiple arbitrary font file includes (?)
* Add a setup/helper function to allow font changes.  I imagine this will work by setting a pointer to the active font array and setting the current height/width which are currently defines, but will be changes to dynamic vars.
* Modify the `oled_write_char` and related functions to support these new font files, making them use the active font array pointer and dynamic height/width.

I _think_ this can all be done without breaking anything which relies on the existing behavior.   And should allow mutiple font sizes on the display at once, memory allowing.  Anyone with a custom font file may have to make some changes (i.e add the width/height bytes to the start), but I've not see to much of this browsing through the user space directories.

brickbots, 2020-03-12:
I've done a bit more digging, and I think that this font file may be a good base:

https://lexus2k.github.io/ssd1306/ssd1306__fonts_8c_source.html

It looks like these are from the GLCD Font Creator program, and have a format very similar to what I was thinking, if you exclude the few non-fixed fonts.    Here is an example of the preamble:

```
0x00, // 0x00 means fixed font type - the only type supported by the library
0x08, // 0x08 = 8 - font width in pixels
0x10, // 0x10 = 16 - font height in pixels
0x20, // Standard filler/delimiter before font data starts
```

Then the font data is a series of bytes from left to right / top to bottom. Unless anyone objects I'll grab this file as a base and see if I can make the required driver changes to support it.

brickbots, 2020-08-23:
Thanks for asking @zigotica.  After my last update I got several fonts setup in the correct format and roughed in some code, but it's not working reliably yet.   I'm sorry to say I've not spent much time on this since then, but your inquiry is good motivation for me to finish this up and get a merge request in.  

Did you have any thoughts about the implementation of this, or just interest in the status and when it might be available?

zigotica, 2020-09-04:
hi @brickbots and thank you. sorry i was out for hols. I didn't get my hands dirty with this just yet. was curious and found out no one had yet tackled this.
pro micro size (ATmega32U4 free space) is an issue for sure, so options might be limited

zigotica, 2020-09-08:
OK, retried using [the 8x16 font](https://lexus2k.github.io/ssd1306/ssd1306__fonts_8c_source.html) suggested by @brickbots above, 

```config.h
#define OLED_FONT_H "keyboards/kyria/keymaps/zigotica/glcdfont.c"
#define OLED_FONT_WIDTH 8
#define OLED_FONT_HEIGHT 16
```

When running `qmk compile` (latest version, I am at cd73949) still complains about it:

```sh
Compiling: drivers/oled/oled_driver.c                                                              drivers/oled/oled_driver.c: In function 'oled_write_char':
drivers/oled/oled_driver.c:388:27: error: 'font' undeclared (first use in this function); did you mean 'float'?
     _Static_assert(sizeof(font) >= ((OLED_FONT_END + 1 - OLED_FONT_START) * OLED_FONT_WIDTH), "OLED_FONT_END references outside array");
                           ^~~~
                           float
drivers/oled/oled_driver.c:388:27: note: each undeclared identifier is reported only once for each function it appears in
drivers/oled/oled_driver.c:388:20: error: expression in static assertion is not an integer
     _Static_assert(sizeof(font) >= ((OLED_FONT_END + 1 - OLED_FONT_START) * OLED_FONT_WIDTH), "OLED_FONT_END references outside array");
                    ^~~~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_kyria_rev1_zigotica/oled_driver.o] Error 1
make: *** [kyria:zigotica] Error 1
Make finished with errors
``` 

Apparently the sizeof is out of range. I think I saw a fix somewhere subtracting one to the size, but I'm not really sure it is 100% related, nor remember the link.

zigotica, 2020-09-08:
OK I _think_ my problem above is related to https://github.com/qmk/qmk_firmware/pull/8145 but that was merged in February

brickbots, 2020-09-08:
Hi @zigotica !  Just to clarify a bit... I've thought about how this might work, and played around a bit on a branch, but I've not committed anything yet.  So, for now I believe only the 8x8 fonts will work out of the box.  

Sorry for any false hope here, I have started working on this (thanks again for pinging about it), but it will probably be a while before it's ready to issue a merge request as I want to try to enable a wider array of fonts without breaking all the other OLED stuff already in play :-)

XScorpion2, 2020-09-09:
A big part of the issue with going to larger font sizes is the structure of the buffer. Currently it's an array where each byte represents 8 pixels vertically, starting top left corner, going right, wrapping to the next row of pixels when we reach the end. 8 bit tall fonts take advantage of this layout by using a memcopy approach copying the 8 bits tall x N bytes wide pixel data directly to the buffer for performance. For increased size fonts, you would need to do 2 passes shifting the bits, and copying 1 byte at a time to the buffer in 2 passes to get the upper and lower bits. Now this could be improved by storing the font data itself differently, separating the upper bytes in one chunk and lower bytes in another chunk. The problem is I haven't done much digging into font creation apps that split the data up like that. The few I have tried instead go horizontal with their bits, so you could do an 8 bits wide, N bytes tall font.

brickbots, 2020-09-09:
Indeed, anything other than 8x8 will definitely be less performant.  I was hoping to get the most flexible, slowest case up and running (arbitrary placement of arbitrary sizes) and see how bad it would be.  Then we could start to maybe set some conditions to speed it up... font sizes must be multiples of 8 high, or only limited to only specific placement in the buffer.

Might ultimately be a fools errand, but it would be nice to have a bit more flexibility on offer if we can make it user-friendly enough.

zigotica, 2020-09-09:
Hey guys, thank you for the insights. I understand it's a hard problem to solve, given the small amount of free space in ATmega32U4 which is the most used micro, I guess. 

> Hi @zigotica ! Just to clarify a bit... I've thought about how this might work, and played around a bit on a branch, but I've not committed anything yet. So, for now I believe only the 8x8 fonts will work out of the box.

No pressure at all @brickbots I understood.

I was also thinking of creating an array and build each char (reduced to alphas and numbers) in 2 or 4 pieces and using a custom method to write each line in 2  runs. Kind of painting a logo but for words (lines actually). That should probably be doable, but again, space is limited. We'll see. 

Thank you again

brickbots, 2020-09-09:
@zigotica That's a great approach, and would definitely work and not require any core changes.  You'd have to come up with a nice font file and a helper function in your keymap, but it's doable and would get you where you want to go.  

Ultimately, it may actually be a good way of handling the bit more general case of double height chars.  It's very much   along the lines of what @XScorpion2 was talking about above with a specially encoded font file that works with the 8 bit memory structure.

XScorpion2, 2020-09-09:
@brickbots @zigotica Yup, that is pretty much what I was suggesting.

zigotica, 2020-09-09:
Great then! I will let you know. Thank you both.

memco, 2021-03-01:
Hi all, I have low vision and would love to be able to use a larger font so I can read the text. Any chance updates are coming soon?

BifbofII, 2021-06-17:
I don't know if anybody is still working on this, but I have started some work on this issue and will create a PR draft soon. What I'm trying to implement is basically what was talked about by @brickbots and @XScorpion2 before, more details in the PR. Feedback is very welcome.

closed 2022-08-28 by zvecr