qmk/qmk_firmware #6658: Not recognized by Windows 10 on boot.

ShizzyGuo, 2019-09-02:
This just recently started happening after the latest Windows 10 update. I have to un plug and re plug my Niu Mini in order to use it whenever I put my pc to sleep or after shutdown/restart. Any one else having this issue? Potential Fix? I compile using the web QMK configurator and flash with the QMK Toolbox

johann-koebbe, 2019-09-06:
I don't have that particular problem, but one of my layers doesn't work on Windows 10 (fully up to date). I've been looking for information, but haven't found any yet. I came here to possibly post an issue. And I did the same, compiled using the online configurator, and flashed with QMK Toolbox 0.0.12.

yanfali, 2019-09-06:
@johann-koebbe please don't hijack other people's issues. If you're having layer problems I would just try the discord and people can help you troubleshoot.

johann-koebbe, 2019-09-06:
@yanfali I wasn't inteding to hijack the issue. I just wanted to add to the conversation about potential problems on Windows. For my own problems, I did create an issue in what I think is the appropriate place. Sorry for the confusion. Just trying to help. :disappointed:

yanfali, 2019-09-06:
@johann-koebbe I've already commented on your issue. Feel free to close your issue once you've tested it. Thanks

drashna, 2019-09-26:
@ShizzyGuo That's very odd.   IIRC, the NIU Mini is using an ATmega32u4 chip, and I don't see this behavior with any other board I'm using. 

If you could, try adding:
```
#define NO_USB_STARTUP_CHECK
```

And see if this helps.

closed 2020-01-24 by stale[bot]

AaronJCA, 2021-01-17:
I am getting the same issue with my NIU Mini. On startup and on waking, the keyboard must be unpluged and pluged back in to work. I haven't tried the #define NO_USB_STARTUP_CHECK suggestion... though I'm not sure how to implement it.