qmk/qmk_firmware #6689: PS/2 Protocol Support

no1so, 2019-09-06:
Hello, I am trying to build a custom keyboard for a project and need it to use the PS/2 protocol instead of USB so that I can interface it with a microcontroller directly. Does your firmware support this/how can I accomplish this?

drashna, 2019-09-26:
You mean so that you're outputting to PS/2 rather than USB?

no1so, 2019-10-06:
Yes, I would like to build a keyboard and have it output via PS/2 protocol so I can interface it with a microcontroller. 

My plan is to have the microcontroller read the keyboard input and run a script to output the characters to a thermal printer. I can do this with my current setup which uses an old PS/2 keyboard, but i'm trying to build a prototype from the ground up with a custom keyboard.

Let me know if you can offer any help. Thanks a bunch!

matheusjgsantos, 2019-12-09:
Looking for the same feature. Lots of new "retro" machine projects based on FPGA are using PS/2 keyboards, so it makes sense having this available on QMK since the demand may increase in the future.

no1so, 2019-12-09:
> Looking for the same feature. Lots of new "retro" machine projects based on FPGA are using PS/2 keyboards, so it makes sense having this available on QMK since the demand may increase in the future.

Yes, what I'm doing now is buying the JJ40 PCB and using ps2avrgb firmware to flash it and I think that should work. But it would be cool to also use the QMK firmware because of its usability.

Let me know if you hear anything about this developing!

closed 2020-04-07 by stale[bot]

neonpixii, 2021-11-24:
would love for this to be supported.  it'd be nice to use my qmk keyboard with a gameboy+lsdj as a control input, which currently only accepts ps/2 input

lebbtheoriginal, 2022-01-19:
interested, my company have a privacy policy so my computer at work has usb disable, would love to use my dactyl at office

zaz, 2025-03-13:
@drashna From what I understand, passive adapters work if the keyboard falls back to PS/2 if the USB handshake fails. Is it feasible to implement this in firmware?

yeroca, 2025-03-13:
> [@drashna](https://github.com/drashna) From what I understand, passive adapters work if the keyboard falls back to PS/2 if the USB handshake fails. Is it feasible to implement this in firmware?

There is a commercial product to do active conversion from USB to PS/2, though they hide the price behind some (stupid) quotation request system: https://www.aten.com/global/en/products/kvm/kvm-modules-%26-accessories/cv10km/

There's also this DIY open source project which does the same thing: https://www.vogons.org/viewtopic.php?t=89777

From what I know about PS/2, what you describe could be done within QMK, provided the USB hardware on the keyboard doesn't block the usage of the D-/D+ pins for other uses, in the PS/2 case it's CLK and DATA.  That said, it wouldn't be  a trivial project to add that capability because the protocol is completely different, and so are the keycodes.  The use cases for PS/2 output are not common and so may not be worth the effort for the core team.  However if you are really ambitious, you could try tackling it yourself with a local QMK tree, and use one of those $5 passive adapters you can get just about anywhere.

If it were me, I'd try getting one of the above converters first just to see how it works out.

fordp2002, 2025-03-14:
There is a large application for retro keyboards for FPGA projects like the Spectrum Next that currently only supports PS/2.

yeroca, 2025-03-15:
> There is a large application for retro keyboards for FPGA projects like the Spectrum Next that currently only supports PS/2.

It's a shame USB requires a much bigger software stack, so I can see why they didn't want that for their Z80-based system.

Out of curiosity, I did a bit of research on the STM32F401chip that's in my keyboard, and found that the USB module can be disabled, and then the GPIO pins normally used for D+/D- can be put into open collector state to be used for a PS/2's CLK and DATA pins.  Then by "bit banging" those GPIO pins, you could produce a PS/2 compatible clock and data.  I also found a conversion table for HID (USB) keycodes and PS/2 keycodes:

https://download.microsoft.com/download/1/6/1/161ba512-40e2-4cc9-843a-923143f3456c/translate.pdf

Without having looked closely at the HID driver code, perhaps the conversion between the HID scan codes and the PS/2 codes could be done at that driver level. 

This is all in extremely sketchy theory, though, and only for the one MCU I checked into.  It would take some work to get all of that working reliably.  One thing for anyone who attempts this is that while bit banging, you'd want to disable interrupts in the MCU in order to keep the CLK rate stably between the 10-15 kHz spec.

Very likely that open source project I mentioned would give a lot of insight how to do the conversion, at least in an isolated environment, and a detailed study of it would expose a lot of considerations and needed behavior that are not obvious from reading the PS/2 spec.

I'm wrote all this here in hopes that it saves someone a little bit of time in the future if they attempt to build this functionality for QMK.