qmk/qmk_firmware #6696: Georgi NKRO fallback mode sends backspace for the ‘I’ brief

fedde-s, 2019-09-07:
I'm running into an issue with my Georgi stenographic keyboard when using it in ‘fake steno/NKRO fallback’ mode. A common steno stroke for the pronoun ‘I’, the chord `RE | RU`, sends a Backspace instead of simulating the chording of N and M on a QWERTY keyboard.

I haven't read the code in too much detail, but I think it might have something to do with the `else` after [this `if` condition](https://github.com/qmk/qmk_firmware/blob/d0c9ad6433d6ed30e8076fd26c8551f579cb08c8/keyboards/georgi/sten.c#L281). It seems to refer back to the same logic as the QWERTY/gaming mode, in which that chord is indeed supposed to send a Backspace.

## System Information

 - Keyboard:  Georgi
   - Revision (if applicable):  Rev2
 - Operating system: reproduced on both Debian GNU/Linux and Android
 - QMK Firmware version: v1.1: ClayMAug  4 2019 (according to the #-TSDZ chord)
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [x] Other: Plover for Linux/Dotterel for Android

fedde-s, 2019-09-07:
I'm sorry if this is not the right place to summarise the issue. @germ, do you see what the problem I'm running into is? And if so, given your familiarity with the code, does a solution spring to mind that doesn't do too much harm to maintainability?

germ, 2019-09-07:
The problem is that the parsing code is calling the main dictionary instead of the proper NKRO fallback when trying to parse out chords. In the mean time, try and use the serial interface with Plover.

I'll have a PR ready for the qmk team in a few hours and link it :)

fedde-s, 2019-09-09:
> In the mean time, try and use the serial interface with Plover.

That mode definitely works great for me so far! And even when it's unavailable, such as on Android, a single-letter word isn't too hard to fingerspell if I remember to use two extra fingers 😅

fedde-s, 2019-09-09:
I later noticed that the NKRO/fake steno fallback mode also sends only the bottommost key, the QWERTY home row key, when I hit a chord consisting solely of two consonant keys above each other (such as PW for “about”). If I understand your diagnosis of the problem correctly, this is the same bug.

fedde-s, 2019-09-16:
@germ, is this still in progress? Is there anything I could help with if I find the time to read up on how to build and load changed code to test it in practice, or do you think that would only make it harder? Not urgently waiting for it, but I'm curious to learn more and contribute.

germ, 2019-09-16:
Oh crap, forgot about this. I don't have my Georgi right now but I'll try hacking at it in the next day or two. I'll ping the Plover discord good chance one of them can fix it up in 5 seconds :)

(Wrote it on my hand so hopefully I don't forget this time)

drashna, 2019-09-26:
May need to update the matrix too :D

zvecr, 2022-08-29:
This issue has been automatically closed because it has not had any recent activity.
If this issue is still valid, re-open the issue and let us know.

closed 2022-08-29 by zvecr

fedde-s, 2022-08-29:
Right, I myself didn't get around to learning QMK and addressing this since several months before The Virus, so I think it would be optimistic to say that it will happen soon :sweat_smile:

fedde-s, 2022-08-29:
Hence I haven't built and loaded the latest code on my device to test whether it persists either.