qmk/qmk_firmware #6704: crkbd makefile error

GlitchOwl, 2019-09-08:
It happens when I use #define USE_MATRIX_I2C in config.h

```
$ make crkbd:default
QMK Firmware 0.7.12
Making crkbd/rev1 with keymap default

avr-gcc.exe (AVR_8_bit_GNU_Toolchain_3.6.1_1752) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   23162       0   23162    5a7a .build/crkbd_rev1_default.hex

Compiling: keyboards/crkbd/ssd1306.c                                                                [OK]
Compiling: keyboards/crkbd/rev1/matrix.c                                                           keyboards/crkbd/rev1/matrix.c: In function 'matrix_scan':
keyboards/crkbd/rev1/matrix.c:246:24: error: 'serial_master_buffer' undeclared (first use in this function)
                (void *)serial_master_buffer, SERIAL_MASTER_BUFFER_LENGTH);
                        ^
keyboards/crkbd/rev1/matrix.c:246:24: note: each undeclared identifier is reported only once for each function it appears in
keyboards/crkbd/rev1/matrix.c:246:46: error: 'SERIAL_MASTER_BUFFER_LENGTH' undeclared (first use in this function)
                (void *)serial_master_buffer, SERIAL_MASTER_BUFFER_LENGTH);
                                              ^
keyboards/crkbd/rev1/matrix.c: In function 'matrix_master_scan':
keyboards/crkbd/rev1/matrix.c:258:9: error: unused variable 'offset' [-Werror=unused-variable]
     int offset = (isLeftHand) ? 0 : ROWS_PER_HAND;
         ^
keyboards/crkbd/rev1/matrix.c:256:9: error: unused variable 'mchanged' [-Werror=unused-variable]
     int mchanged = 1;
         ^
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:380: .build/obj_crkbd_rev1_default/matrix.o] Error 1
Make finished with errors
make: *** [Makefile:561: crkbd:default] Error 1
```

yanfali, 2019-09-09:
@GlitchOwl pretty crkbd is hard coded to only support serial for the keyboard matrices because it uses the i2c pins for the OLED per side.

drashna, 2019-09-09:
It's based on helix code.  But I'm not sure that either really have been tested with i2c for the split communication, because both are designed to use serial for communication and i2c for OLEDs.  And enabling i2c for split communication requires adding pull up resistors, actually.  And you cannot use OLEDs on both sides, due to addressing conflicts, in this case. 

I think you need to remove the define for `USE_SERIAL` in this case.

drashna, 2019-11-03:
Was this still an issue?

closed 2020-03-02 by stale[bot]