qmk/qmk_firmware #6705: Shift != SFTENT; Controlling the Shift+Key behaviour

rodrigob, 2019-09-08:
(cross posting from [reddit](https://www.reddit.com/r/olkb/comments/d1d1ci/shift_sftent_controlling_the_shiftkey_behaviour/))

I have remapped both Shift and Enter keys to behave as KC_SFTENT.
Similarly I have remapped the Backspace to behave as Delete when doing Shift+Backspace.
A problem that I have encountered is that some features in QMK do not seem to combine well together.

In particular the technique from [this post](
https://www.reddit.com/r/olkb/comments/cylzth/qmk_can_i_change_the_behavior_of_a_shifted_key/eyts203?utm_source=share&utm_medium=web2x) does not combine with KC_SFTENT.
I also tried the code from [this fork](
https://github.com/qmk/qmk_firmware/pull/4795) but that did not work for me (shift+key seemed to do nothing).

What is the recommended way to approach to handle this?

I cannot believe I am the first one trying to use & abuse the Shift and Backspace keys.
Suggestions and ideas most welcome!

ps: two additional recent threads on the Shift+Key topics:
https://www.reddit.com/r/olkb/comments/cylzth/qmk_can_i_change_the_behavior_of_a_shifted_key/
https://www.reddit.com/r/olkb/comments/czyo9b/help_qmk_send_different_key_when_shift_is_pressed/

fauxpark, 2019-09-09:
I can't seem to replicate this; tapping `KC_DTQS` results in `.`, tapping it with `KC_SFTENT` held produces `?` as expected.

What keyboard(s) are you seeing this on?

rodrigob, 2019-09-09:
Thanks for the reply.
I am using a pinky4 with the following [`keymap.c`](https://gist.github.com/rodrigob/e220614d4be9dc373a2997ef8ec153a7).
`rules.mk` has `EXTRAKEY_ENABLE = yes` and everything else set as `no`.

rodrigob, 2019-09-10:
Is there any other debug information I can/should provide ?

fauxpark, 2019-09-10:
Does it work if both keys are on the same (master) half?
Also, the default keymap for the pinky includes some defines that change the behaviour of tap/hold keys in its config.h - if you just copied the entire folder, one of those might be the culprit.

AdyaAdya, 2019-09-14:
Hello,

I'm the originator of this [pull request](https://github.com/qmk/qmk_firmware/pull/2900) which is now closed, that involved controlling keycodes sent to the host depending on the Shift and RightAlt modifiers' state. (I called that feature Custom Modified Values or CMV)

The [pull request](https://github.com/qmk/qmk_firmware/pull/4795) that you linked is the version of the feature that another user shared. The main difference between their implementation and mine is that their code remains mostly in user-level code and does not go too deep in the firmware code, while my code does. This is the reason why their code has been kept to represent the feature, as it is "cleaner" in a certain way compared to mine.

However, if I understand correctly your issue, your problem comes from the fact that the keycodes your trying to send when shifted are special keys that require being processed by the firmware just as if it were a regular key press. When I implemented my version of the feature, I tried to make that work and support every weird QMK keycodes. So, if you try my implementation or use it as a template to make your own, you might be able to solve your problem.

Here's my fork (latest commit on branch master is the newest version): [https://github.com/AdyaAdya/qmk_firmware](https://github.com/AdyaAdya/qmk_firmware)

Full disclaimer, I never took the time to test every keycodes and features of QMK firmware to see if my implementation of CMV was compatible with them, so I can't assure you that it'll work.

closed 2022-08-03 by tzarc