qmk/qmk_firmware #6753: Planck EZ not tresponsive after flashing

DarkDeepBlue, 2019-09-16:
After having fetched the newest master and having flashed my Planck EZ, it has become not responsive. If I press a key now, it is recognized only a few seconds later. Same with releasing the key.

## Describe the Bug

I have updated the qmk firmware from version 0.6.441 to 0.7.15 and have flashed the keyboard. Now, it has a huge delay registering the keys. After pressing a key, it takes about three seconds until it gets recognized. Releasing the key has the same delay. Switching to different versions including the previously working version, 0.6.441, has not lead to success.

Also, using the qmk configurator one can retrieve a binary independent of particular compiler versions. Currently such a file does not lead to success after flashing, either. The keymap does change, but the delay remains.

## System Information

 - Keyboard:  Planck
   - Revision (if applicable):  EZ
 - Operating system: ArchLinux
 - AVR GCC version: 9.2.0
 - ARM GCC version: 9.2.0
 - QMK Firmware version: 0.7.15
 - Any keyboard related software installed? No.

## Additional Context

Ergodox EZ works flawlessly with current master.

If necessary, I can reproduce the file that has been flashed on the keyboard.

I am stilll able to set the keyboard in DFU bootloader mode and flash it over again. The issue seems to be caused from something that does not get fixed by flashing a clean file over again.

Can #6609 explain such behavior? Otherwise I have not found anything related in the existing issues. Does anyone have any ideas in this case?

yanfali, 2019-09-17:
@DarkDeepBlue we don't support your GCC version. Please try 8.3 (AVR or ARM) or 6.3 (ARM). The 9 series is deeply broken for us at the moment. If you get it to work I would close this issue. Thanks

yanfali, 2019-09-18:
@DarkDeepBlue made any progress? please update and/or close this issue.

DarkDeepBlue, 2019-09-18:
I was unable to recover.

There are two main lessons that can be derived from this. Firstly, I would use the provided docker environment to build the keymaps. I suspect that the compiler versions I used are responsible for the initial issue. The docker environment provides compiler versions that are stable and are known to provide correct code. Secondly, whatever  you do, do not overwrite the optional bytes of the `STM32F303` MCU.

closed 2019-09-18 by DarkDeepBlue

DarkDeepBlue, 2019-09-26:
There is an update on this issue.

I have investigated the cause for the issue more using `hid_listen` and `uprint()`, focusing on the matrix scanning logic. It seems that the performance issue is caused by the call to ``

DarkDeepBlue, 2019-09-26:
I have investigated the issue deeper and would like to share my results in case others face similar performance issues.

The performance regression seems to be caused by the call to `rgb_matrix_task()` in `quantum/quantum.c`, line `1010` of the current `master` (`d143ddc06`). One can disable the RGB matrix scan by adding `RGB_MATRIX_ENABLE = no` to the `rules.mk`. Then, the corresponding scan is skipped, resulting in far better performance during booting and running the firmware.

I am using a board without RBG lights, so this is fine for me. Others might want to look into the issue more closely in case this is not an isolated case.

Also, take note that this issue does not seem to be caused by specific compiler versions, as this time I have used the versions provided in the `docker` environment.

uqs, 2021-08-11:
THANK YOU SO MUCH for the pointer to RGB being at fault. I have a Preonic and it would be dead with every custom compiled firmware that I produced.

While investigating downgrading GCC a la https://github.com/qmk/qmk_firmware/issues/5868 I noticed your report, and while preonic/rev3 doesn't set `RGB_MATRIX_ENABLE`, it sets `RGBLIGHT_ENABLE`.

Guess what? Disabling both results in GCC 11 producing a firmware that actually boots up the board. Hurray!